KaufHA/kauf-rgbww-bulbs #1: How do I reflash a bricked bulb?

didymus, 2022-01-01:
I've popped off the diffuser. All I see are the LEDs. I removed the 2 screws and I just see a bunch of thermal paste and a battery down in the neck. Is there something else down in the neck that I need to connect the FTDI to? I have no idea how these are constructed so I'd like to get a little more information first. I can't find anything on the internet. I sent you an email and then there is this. Any information would be appreciated. Am I supposed to be looking at one of the tiny chips on the LED board?

KaufHA, 2022-01-01:
You've gone too far.  You only have to take off the diffuser.  Screw the PCB back down onto the substrate with the thermal paste.  Here's a note I sent someone else with instructions.  I've been meaning to make a video.

--------------------

Note: Do not have the bulb plugged into a socket when trying this.  The bulb should not be powered by 120v.  It will get 3.3v from the FTDI adapter.  Also make sure the FTDI adapter is configured for 3.3v.   They usually have a jumper to select between 5v power and 3.3v power.

The main challenge is that the pads on the bulbs are so small to solder to.  For that reason, I recommend that you use as thin of wire as possible.  The thinner wire is easier to solder on and it allows the wire to flex a lot without applying too much pressure to the pads which could damage them.

Here is how I do mine:

[image]

The green wire is to connect GPIO0 to ground and then allow it to be disconnected so that the ESP8266 can boot up normally when being powered by the FTDI.

To open the bulb, just use anything grippy to pull the two parts together.  I use two pieces of kitchen shelf liner and they pop apart pretty easily.  With the BR30's, it might be a bit tougher to pull apart.  Twisting works pretty well for me, but you may need to pry apart with a screwdriver or something similar.

The 5 pads on the bulb's PCB that you have to connect to are GND, RX, TX, 3V3 (vcc) and IO0.  Here is a picture with the pads numbered 1-5:
[image]

You have to connect VCC, GND, RX, and TX to the FTDI adapter.
GND -> GND
VCC -> 3V3
RX -> TX
TX -> RX

IO0 from the bulb does not connect to the FTDI adapter.  It just connects to ground while the FTDI adapter is plugged into USB.  You can have IO0 hard wired to ground just to flash, it doesn't need to be switchable like on mine.

Once you have all 5 terminals connected as described above, plug into USB and try to flash.  I use the program FlashESP8266.exe that comes in this package (https://github.com/letscontrolit/ESPEasy/releases/download/mega-20211105/ESPEasy_ESP82xx_mega-20211105.zip).  You could also use ESPHome Flasher (https://github.com/esphome/esphome-flasher/releases/tag/1.4.0).

bkaufx, 2022-10-11:
This 3d printable jig may be helpful: https://www.thingiverse.com/thing:5269303