KaufHA/kauf-rgbww-bulbs #2: Issues with OTA (v1.72)

Exit2Studios, 2022-01-13:
I am getting the following error in esphome when trying to validate or update OTA my new bulbs that I updated to v1.72 firmware:

INFO Reading configuration /config/esphome/kauf-bulb-d5f126.yaml...
INFO Updating https://github.com/KaufHA/kauf-rgbww-bulbs@None
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
ERROR Unable to import component web_server:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/.esphome/external_components/69101fc3/components/web_server/__init__.py", line 5, in <module>
    from esphome.const import (
ImportError: cannot import name 'CONF_INCLUDE_INTERNAL' from 'esphome.const' (/esphome/esphome/const.py)
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
Failed config

button: [source /config/esphome/kauf-bulb-d5f126.yaml:58]
  
  Component not found: button.
  - platform: restart
    id: restart_button
    name: Office Floor Lamp Restart Firmware
    entity_category: diagnostic
    disabled_by_default: True
web_server: [source /config/esphome/kauf-bulb-d5f126.yaml:99]
  
  Component not found: web_server.

KaufHA, 2022-01-13:
Do you have ESPHome fully updated?  What version is it?  Buttons were only added in the latest major release (2021.12.0) so it seems like maybe you are on an older version that doesn't have buttons.  That could easily be causing the web_server issue as well.

You can check the version number at the bottom of the ESPHome dashboard or in the Home Assistant add-on configuration page:

[image]

Exit2Studios, 2022-01-13:
That is strange.  I am on version 2021.11.3, but don't have any option to update.

KaufHA, 2022-01-13:
Yea, that is weird.  Are you using Home Assistant OS with the supervisor and ESPHome installed as an add-on?  Maybe you need to update the OS?  I don't know enough about the supervisor to know why you wouldn't be seeing an update when one is available.  My OS install showed the ESPHome update just fine.

What version of Home Assistant are you running, and is it showing an update available?  The latest should be 2021.12.8.  I'm wondering if your supervisor is stuck and not showing any updates, or if this is just an ESPHome issue.

Looks like you aren't the only one with this problem: https://community.home-assistant.io/t/esphome-stuck-on-update-2021-11-1/368560/6.  Maybe try restoring a backup of the ESPHome add-on and then upgrading from there.

Exit2Studios, 2022-01-13:
I am on the latest version (2021.12.9), but have no update available for the ESPHome add-on (I am running supervior).

Exit2Studios, 2022-01-13:
I don't think I have a backup that is old enough available.

Exit2Studios, 2022-01-13:
So I solved my own problem, but I'll post here for prosperity sake and hopefully in the case that it helps someone else.  I went to the add-on store and installed the latest version of ESPHome.  I ended up with two separate versions/add-ons.  All of my devices showed up on the new one, so I simply stopped the other and will delete once I confirm it is not needed.

KaufHA, 2022-01-13:
Glad you got it.  The other thing I was going to recommend would be to uninstall the add-on and reinstall.  You would just need to back up all the yaml files manually and then copy them back into the ESPHome config directory after reinstalling.

closed 2022-01-13 by KaufHA