KaufHA/kauf-rgbww-bulbs #3: ESPHome compile failed on stock yaml file with new components

wifijt, 2022-01-20:
Just tried to flash a bulb using the stock YAML file and got this error:

`In file included from src/esphome/components/restart/button/restart_button.h:3,
                 from src/esphome/components/restart/button/restart_button.cpp:1:
src/esphome/components/restart/button/restart_button.cpp: In member function 'virtual void esphome::restart::RestartButton::dump_config()':
src/esphome/components/button/button.h:11:3: warning: 'nonnull' argument 'this' compared to NULL [-Wnonnull-compare]
   11 |   if ((obj) != nullptr) { \
      |   ^~
src/esphome/components/restart/button/restart_button.cpp:17:37: note: in expansion of macro 'LOG_BUTTON'
   17 | void RestartButton::dump_config() { LOG_BUTTON("", "Restart Button", this); }
      |                                     ^~~~~~~~~~
In file included from src/esphome/components/template/select/template_select.h:3,
                 from src/esphome/components/template/select/template_select.cpp:1:
src/esphome/components/template/select/template_select.cpp: In member function 'virtual void esphome::template_::TemplateSelect::dump_config()':
src/esphome/components/select/select.h:13:3: warning: 'nonnull' argument 'this' compared to NULL [-Wnonnull-compare]
   13 |   if ((obj) != nullptr) { \
      |   ^~
src/esphome/components/template/select/template_select.cpp:64:3: note: in expansion of macro 'LOG_SELECT'
   64 |   LOG_SELECT("", "Template Select", this);
      |   ^~~~~~~~~~
Compiling .pioenvs/erikas-desk-bulb/src/esphome/components/web_server/web_server.cpp.o
Compiling .pioenvs/erikas-desk-bulb/src/esphome/components/web_server_base/web_server_base.cpp.o
In file included from src/esphome/components/uptime/uptime_sensor.h:4,
                 from src/esphome/components/uptime/uptime_sensor.cpp:1:
src/esphome/components/uptime/uptime_sensor.cpp: In member function 'virtual void esphome::uptime::UptimeSensor::dump_config()':
src/esphome/components/sensor/sensor.h:13:3: warning: 'nonnull' argument 'this' compared to NULL [-Wnonnull-compare]
   13 |   if ((obj) != nullptr) { \
      |   ^~
src/esphome/components/uptime/uptime_sensor.cpp:31:36: note: in expansion of macro 'LOG_SENSOR'
   31 | void UptimeSensor::dump_config() { LOG_SENSOR("", "Uptime Sensor", this); }
      |                                    ^~~~~~~~~~
Compiling .pioenvs/erikas-desk-bulb/src/esphome/components/wifi/wifi_component.cpp.o
src/esphome/components/light/light_json_schema.cpp: In static member function 'static void esphome::light::LightJSONSchema::dump_json(esphome::light::LightState&, ArduinoJson::JsonObject)':
src/esphome/components/light/light_json_schema.cpp:55:53: error: use of deleted function 'ArduinoJson::JsonObject::JsonObject(const ArduinoJson::JsonObject&)'
   55 |   JsonObject color = root.createNestedObject("color");
      |                                                     ^
In file included from .piolibdeps/erikas-desk-bulb/ArduinoJson-esphomelib/src/ArduinoJson.hpp:9,
                 from .piolibdeps/erikas-desk-bulb/ArduinoJson-esphomelib/src/ArduinoJson.h:9,
                 from src/esphome/components/json/json_util.h:8,
                 from src/esphome/components/light/light_json_schema.h:7,
                 from src/esphome/components/light/light_json_schema.cpp:1:
.piolibdeps/erikas-desk-bulb/ArduinoJson-esphomelib/src/ArduinoJson/JsonObject.hpp:40:7: note: 'ArduinoJson::JsonObject::JsonObject(const ArduinoJson::JsonObject&)' is implicitly deleted because the default definition would be ill-formed:
   40 | class JsonObject : public Internals::JsonPrintable<JsonObject>,
      |       ^~~~~~~~~~
.piolibdeps/erikas-desk-bulb/ArduinoJson-esphomelib/src/ArduinoJson/JsonObject.hpp:40:7: error: 'ArduinoJson::Internals::NonCopyable::NonCopyable(const ArduinoJson::Internals::NonCopyable&)' is private within this context
In file included from .piolibdeps/erikas-desk-bulb/ArduinoJson-esphomelib/src/ArduinoJson/Deserialization/../JsonBuffer.hpp:11,
                 from .piolibdeps/erikas-desk-bulb/ArduinoJson-esphomelib/src/ArduinoJson/Deserialization/JsonParser.hpp:7,
                 from .piolibdeps/erikas-desk-bulb/ArduinoJson-esphomelib/src/ArduinoJson/JsonBufferBase.hpp:7,
                 from .piolibdeps/erikas-desk-bulb/ArduinoJson-esphomelib/src/ArduinoJson/DynamicJsonBuffer.hpp:7,
                 from .piolibdeps/erikas-desk-bulb/ArduinoJson-esphomelib/src/ArduinoJson.hpp:7,
                 from .piolibdeps/erikas-desk-bulb/ArduinoJson-esphomelib/src/ArduinoJson.h:9,
                 from src/esphome/components/json/json_util.h:8,
                 from src/esphome/components/light/light_json_schema.h:7,
                 from src/esphome/components/light/light_json_schema.cpp:1:
.piolibdeps/erikas-desk-bulb/ArduinoJson-esphomelib/src/ArduinoJson/Deserialization/../Data/NonCopyable.hpp:17:3: note: declared private here
   17 |   NonCopyable(const NonCopyable&);
      |   ^~~~~~~~~~~
src/esphome/components/light/light_json_schema.cpp: In static member function 'static void esphome::light::LightJSONSchema::parse_color_json(esphome::light::LightState&, esphome::light::LightCall&, ArduinoJson::JsonObject)':
src/esphome/components/light/light_json_schema.cpp:98:36: error: use of deleted function 'ArduinoJson::JsonObject::JsonObject(const ArduinoJson::JsonObject&)'
   98 |     JsonObject color = root["color"];
      |                                    ^
In file included from .piolibdeps/erikas-desk-bulb/ArduinoJson-esphomelib/src/ArduinoJson/Deserialization/../JsonVariantBase.hpp:7,
                 from .piolibdeps/erikas-desk-bulb/ArduinoJson-esphomelib/src/ArduinoJson/Deserialization/../JsonVariant.hpp:13,
                 from .piolibdeps/erikas-desk-bulb/ArduinoJson-esphomelib/src/ArduinoJson/Deserialization/../JsonBuffer.hpp:12,
                 from .piolibdeps/erikas-desk-bulb/ArduinoJson-esphomelib/src/ArduinoJson/Deserialization/JsonParser.hpp:7,
                 from .piolibdeps/erikas-desk-bulb/ArduinoJson-esphomelib/src/ArduinoJson/JsonBufferBase.hpp:7,
                 from .piolibdeps/erikas-desk-bulb/ArduinoJson-esphomelib/src/ArduinoJson/DynamicJsonBuffer.hpp:7,
                 from .piolibdeps/erikas-desk-bulb/ArduinoJson-esphomelib/src/ArduinoJson.hpp:7,
                 from .piolibdeps/erikas-desk-bulb/ArduinoJson-esphomelib/src/ArduinoJson.h:9,
                 from src/esphome/components/json/json_util.h:8,
[... 127 lines cut ...]
src/esphome/components/json/json_util.h:20:13: note:   candidate expects 2 arguments, 1 provided
src/esphome/components/json/json_util.h:22:13: note: candidate: 'std::string esphome::json::build_json(const json_build_t&)'
   22 | std::string build_json(const json_build_t &f);
      |             ^~~~~~~~~~
src/esphome/components/json/json_util.h:22:44: note:   no known conversion for argument 1 from 'esphome::web_server::WebServer::light_json(esphome::light::LightState*)::<lambda(ArduinoJson::JsonObject)>' to 'const json_build_t&' {aka 'const std::function<void(ArduinoJson::JsonObject&)>&'}
   22 | std::string build_json(const json_build_t &f);
      |                        ~~~~~~~~~~~~~~~~~~~~^
src/esphome/components/web_server/web_server.cpp: In member function 'std::string esphome::web_server::WebServer::select_json(esphome::select::Select*, const string&)':
src/esphome/components/web_server/web_server.cpp:755:4: error: no matching function for call to 'build_json(esphome::web_server::WebServer::select_json(esphome::select::Select*, const string&)::<lambda(ArduinoJson::JsonObject)>)'
  755 |   });
      |    ^
In file included from src/esphome/components/web_server/web_server.cpp:8:
src/esphome/components/json/json_util.h:20:13: note: candidate: 'const char* esphome::json::build_json(const json_build_t&, size_t*)'
   20 | const char *build_json(const json_build_t &f, size_t *length);
      |             ^~~~~~~~~~
src/esphome/components/json/json_util.h:20:13: note:   candidate expects 2 arguments, 1 provided
src/esphome/components/json/json_util.h:22:13: note: candidate: 'std::string esphome::json::build_json(const json_build_t&)'
   22 | std::string build_json(const json_build_t &f);
      |             ^~~~~~~~~~
src/esphome/components/json/json_util.h:22:44: note:   no known conversion for argument 1 from 'esphome::web_server::WebServer::select_json(esphome::select::Select*, const string&)::<lambda(ArduinoJson::JsonObject)>' to 'const json_build_t&' {aka 'const std::function<void(ArduinoJson::JsonObject&)>&'}
   22 | std::string build_json(const json_build_t &f);
      |                        ~~~~~~~~~~~~~~~~~~~~^
*** [.pioenvs/erikas-desk-bulb/src/esphome/components/web_server/web_server.cpp.o] Error 1
========================= [FAILED] Took 52.18 seconds =========================
`

wifijt, 2022-01-20:
Here is the YAML file:

`substitutions:

  name: erikas-desk-bulb              # **** CHANGE DEVICE NAME TO SOMETHING UNIQUE PER DEVICE.  RENAME YAML FILE TO SAME NAME.    ****
                               # **** USE DASHES (-) INSTEAD OF SPACES OR UNDERSCORE (_).  USE ONLY LOWER CASE LETTERS.     ****
  
  friendly_name: Erikas Desk Bulb     # **** CHANGE FRIENDLY NAME TO SOMETHING UNIQUE PER DEVICE ****

esp8266:                       # https://esphome.io/components/esp8266.html
  board: esp01_1m
  restore_from_flash: true

  framework:                   # provides updated PWM logic per https://github.com/esp8266/Arduino/pull/7231
    version: 3.0.2             # https://github.com/esp8266/Arduino/releases
    platform_version: 3.2.0    # https://github.com/platformio/platform-espressif8266/releases/

external_components:
  - source:
      type: git
      url: https://github.com/KaufHA/kauf-rgbww-bulbs
    refresh: 0s
    
    # kauf_rgbww and light are required external components.  If you don't want to use them, you should use kauf-bulb-minimal.yaml instead of kauf-bulb.yaml
    # web_server is not required, it just adds the version number to the web interface and also a warning about not flashing WLED or tasmota-minimal.
    # web_server_base is not required, it just makes the OTA system error out if wled or minimal are found in an uploaded firmware filename.
    components: [kauf_rgbww,light,web_server,web_server_base]
    
esphome:

  name: $name

  project:
    name: "kauf.rgbww"
    version: "1.72"

  on_boot:
    then:

      - lambda: |-

          // restore light state per setting.  Default is to restore previous state so we don't need an action for that.
          if (id(select_boot_state).state == "Always On - Last Value") {
            auto call = id(kauf_light).turn_on();
            call.perform();
          }
          else if (id(select_boot_state).state == "Always On - Bright White") {
            auto call = id(kauf_light).turn_on();
            call.set_color_mode(ColorMode::COLOR_TEMPERATURE);
            call.set_color_temperature(150);
            call.set_brightness(1.0);
            call.perform();
          }
          else if (id(select_boot_state).state == "Always Off") {
            auto call = id(kauf_light).turn_off();
            call.perform();
          }

          // turn on WLED if effect is selected
          if ( id(effect).state == "WLED / DDP" ) { id(kauf_light)->set_use_wled(); }

button:
  - platform: restart
    id: restart_button
    name: $friendly_name Restart Firmware
    entity_category: diagnostic
    disabled_by_default: true

wifi:

[... 128 lines cut ...]
      - script.execute: save_changes
      
    entity_category: config

sensor:
  - platform: uptime
    name: $friendly_name Uptime
    update_interval: 60s
    entity_category: diagnostic
    disabled_by_default: true

# Send IP Address to HA.  Disabled by default, not as important now that Home Assistant provides link.
text_sensor:
  - platform: wifi_info
    ip_address:
      name: $friendly_name IP Address
      entity_category: diagnostic
      disabled_by_default: true

script:
  - id: save_changes
    mode: restart
    then:
      - delay: 3s
      - lambda: 'global_preferences->sync();'`

KaufHA, 2022-01-20:
What version of ESPHome are you running?  This looks like you need to upgrade to the latest version 2022.1.1.

BTW, I'm in the process of updating documentation and recommendations to use automatic importing of devices into the dashboard.  Here is what the yaml file would look like for that.  You can copy and paste this over your existing yaml and change names etc.   This will cause any updates on our end to be automatically incorporated.

```
substitutions:
  name: bedroom-lamp-1
  friendly_name: Bedroom Lamp 1

packages:
  kauf.rgbww: github://KaufHA/kauf-rgbww-bulbs/kauf-bulb.yaml

esphome:
  name_add_mac_suffix: false

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  
  manual_ip:
    static_ip: !secret bedroom_lamp_1_ip_address
    gateway: !secret wifi_gateway
    subnet: !secret wifi_subnet
    dns1: !secret wifi_dns1

```

wifijt, 2022-01-20:
Excellent - I will upgrade and retry - love the new YAML file and the automatic inclusions with the packages line. 

Is there a way to have it error out if ESPHome is not a certain version?

KaufHA, 2022-01-20:
I was just thinking about how to error out if version number is wrong.  It should be possible. I will try to figure it out.

KaufHA, 2022-03-17:
I added an error if you aren't on the right version of ESPHome.  Looks like this:

```

#if ESPHOME_VERSION_CODE < VERSION_CODE(2022, 3, 0)
  #error "Please Update ESPHome to the latest version.  Expecting 2022.3.X"
#endif

#if ESPHOME_VERSION_CODE >= VERSION_CODE(2022, 4, 0)
  #error "KAUF external components have not been updated for this version of ESPHome yet, or you are not using the latest KAUF external components version.  Expecting 2022.3.X.  You can try deleting the .esphome/packages and .esphome/external_components subdirectories within the ESPHome config directory to resolve this."
#endif
```

closed 2022-03-17 by KaufHA