KaufHA/kauf-rgbww-bulbs #4: WLED functionality for other bulbs.

jonadis, 2022-01-24:
Greetings,

First off let me say these bulbs are fantastic, and I'm installing them all over my house.
That being said I do have some existing lights running ESPHome using just basic stock rgbw functionality.
I'd replace these too but they are hard to get to and/or are a form factor that Kauf doesn't currently sell a product for.
I see the Kauf bulbs support WLED/xLights (DDP) functionality and I tried that out and it works well.  My question is, could that functionality ported to be used on other RGBW products?  I have some that the same PWM style as teeh Kauf, but I also have some that are MY92X1 and some that are sm16716.  I'm just wondering if that's possible with stock ESPHome or if the Kauf custom files here on GitHub are required, and if its the latter if it would be possible to pull that into stock ESPHome in a future release?

Keep up the good work!

KaufHA, 2022-01-24:
You need our custom light component.  I think you could do it for any RGB bulb and it wouldn't be too hard.  We didn't put it together with that in mind, so I can't promise there won't be some weird issues.  Just look at how we add external components in our yaml to pull in the custom light component.  You could also take the effect select entity from our yaml to enable/disable.

I'd love to get this working in stock ESPHome but I haven't had time to learn how their effects system works and I've also never done a pull request like that so there is a bit of a learning curve I'm on.  It's on my list of things to do but I have been pretty swamped with other things.

bkaufx, 2022-12-14:
I went ahead and created a ddp component that should work for any light:

https://github.com/KaufHA/common/tree/main/components#ddp