KaufHA/kauf-rgbww-bulbs #5: Bulbs getting Adopted after already adopting

Exit2Studios, 2022-02-10:
I've gone through and updated the firmware on some bulbs that shipped pre 1.72.  I adopted them into esphome and changed the config, renamed the yaml file, etc.  However, I'm getting the following for each of those bulbs (and unable to adopt if I try).  How do I get this to go away?
[image]

KaufHA, 2022-02-10:
Did you compile the firmware and flash to the bulb?  If not, then the old firmware is still running and being found in addition to the yaml file.

Here's what I have when I adopt but don't install:

[image] [image]

The green one is shown due to the firmware running on the network broadcasting its dashboard import capability.  The red one is due to a yaml file in the config directory.  It's red because there is no corresponding device found on the network.  Only once I "update" the adopted bulb then ESPHome doesn't see the old firmware available to adopt.

If you did already flash new firmware to the bulb then I'd try deleting the ".esphome" folder within the esphome config directory.  Maybe something is cached funny.

KaufHA, 2022-02-10:
Maybe restart the esphome addon or dashboard as well when you delete .esphome

closed 2022-11-30 by KaufHA