KaufHA/kauf-rgbww-bulbs #8: ESPhome error with new firmware

Zed222, 2022-03-26:
When trying to adopt in or update in ESPhome I get the following. Also of note I did load plug firmware mistakenly. ESP home still finds both plug and bulb.

INFO Reading configuration /config/esphome/kauf-bulb-5.yaml...
INFO Updating https://github.com/KaufHA/common@None
INFO Updating https://github.com/KaufHA/kauf-rgbww-bulbs@None
ERROR Unable to import component switch:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/.esphome/external_components/4d68583c/components/switch/__init__.py", line 6, in <module>
    from esphome.const import (
ImportError: cannot import name 'DEVICE_CLASS_OUTLET' from 'esphome.const' (/esphome/esphome/const.py)
Failed config

switch: [source /config/esphome/.esphome/packages/0cf47246/kauf-bulb.yaml:320]
  
  Component not found: switch.
  - platform: template
    id: switch_no_hass
    name: Kauf Bulb 5 No HASS
    optimistic: True
    restore_state: True
    entity_category: config
    disabled_by_default: true
    icon: mdi:toggle-switch-off-outline
    forced_hash: 706743398
    forced_addr: 70
    global_addr: global_forced_addr
    on_turn_on: 
      - script.execute: script_no_hass
      - script.execute: script_save_changes
    on_turn_off: 
      - script.execute: script_no_hass
      - script.execute: script_save_changes

KaufHA, 2022-03-26:
What version of ESPHome do you have?  Kind of looks like you aren't upgraded to the latest (2022.3.1).  The other thing that commonly resolves this type of error is to delete the .esphome/packages and .esphome/external_compenents subfolders within the ESPHome config directory.  Let me know if those ideas don't work.

Zed222, 2022-03-26:
Thanks. I will update and than move on to deleting suggested files.

Zed222, 2022-03-26:
Updated HA to v7.5 and ESPHome to v2022.3.1. Did NOT delete the files. Was able to update and rename device in ESPHome. However HA does not seem to recognize the device so no entities to integrate.

closed 2022-03-26 by Zed222

KaufHA, 2022-03-26:
Probably need to remove the old device from the ESPHome integrations card (named kauf-bulb-xxxxx).  Then either restart the bulb and HA should pick it up or you can manually add by IP address as shown here: https://kaufha.com/troubleshooting/

Zed222, 2022-03-26:
Success !

Thank You for the help. Been following your videos on  YouTube, they have been a great resource. Look forward to the new products.