KaufHA/kauf-rgbww-bulbs #10: Power On State clarification

Cossid, 2022-04-03:
I just updated to the latest firmware 1.83(y), and the Power On State has both "Restore Power Off State" and "YAML Configured (RESTORE_DEFAULT_OFF)".

I'm assuming these are the same?  Code for "Restore Power Off State" appears to point to RESTORE_DEFAULT_OFF, just curious why there are two records for this?

To be clear, I haven't entered into the device config yaml myself, I'm only using the 10 line yaml from the README.md

KaufHA, 2022-04-03:
The base kauf-bulb.yaml that gets imported as a package has the substitution, and therefore so do you even without explicitly defining it locally.  The substitution can be used to change the restore mode to any option you want in your local yaml (if you add the substitution locally it will overwrite the value from kauf-bulb.yaml).  However, the way the select entity is implemented would overwrite the substitution if I didn't have an option in the select entity to let the substitution rule.

By default, the base yaml sets the substitution to RESTORE_DEFAULT_OFF, which is indeed the same as "Restore Power Off State".   The update firmware bin file is generated with the substitution overwritten to ALWAYS_ON which is equivalent to the option "Always on - Bright White".  The select entity option allows different defaults for the dashboard import and update/factory firmwares by modifying the substitution.

Cossid, 2022-04-03:
That makes perfect sense, thanks for the explanation.  That might be useful information to the README.md (what YAML Configured means, it's currently missing from the list), as by default it isn't very clear why there are two.

closed 2022-04-03 by Cossid