KaufHA/kauf-rgbww-bulbs #9: Suggestion for out-of-box experience

Cossid, 2022-04-03:
Just received my first 4 pack of the Kauf BR30 and there are a couple things that would make the out-of-box experience a little better.

It would be nice if the ssid selection screen also included:

- the current host name (it is present in the title of the page, but cannot be copied to clipboard)
- the wifi mac address (which makes it easier to locate in the dhcp listings)

They would be useful on the default screen after it has joined the wifi network as well.

Additionally, you already have a HA diagnostic entry for IP address, having one for mac address would also be useful, for setting static dhcp entries.

Lastly, while I understand these are meant to be used with HA (latest firmware has a "No HASS" option), it would be nice to have an interface to control color/temperature from the device's own UI as a last resort.  I understand if this one is not in the cards.

KaufHA, 2022-04-03:
Great suggestions, thanks.  Just had someone else make some similar suggestions (more info on captive portal / web interface pages) so I'll add these to his and make them a priority.  As far as controlling color etc. from the web interface, I might look into it.  That's the kind of functionality I'm just relying on base ESPHome for, but maybe I can do a pull request and contribute back to the project if it seems like something I could do reasonably easily.

If you import the bulbs into the ESPHome dashboard, you could add the mac address yourself by adding these lines:

```
text_sensor:
  - platform: wifi_info
    mac_address:
      name: ESP Mac Wifi Address
```

https://esphome.io/components/text_sensor/wifi_info.html

Cossid, 2022-04-03:
Thanks for the quick info.  The only reason I was looking for color control on the ESP page would have been to do a quick diagnostic check before I added them in HA, so don't spend too much time working on a color control scheme in ESPHome (I'm surprised they don't have something stock in ESPHome honestly), it would just be a nicety to help with user friendlyness.

I'll leave this issue open solely for the hostname/mac address display requests, and you can close it when you add it.

KaufHA, 2022-11-30:
Host name and mac address are now displayed at the bottom of the captive portal and normal web server.

closed 2022-11-30 by KaufHA