KaufHA/kauf-rgbww-bulbs #11: Error upgrading

Exit2Studios, 2022-04-26:
I'm getting the following error when trying to update in esphome.  Running esphome v2022.4.0.  I have 3 bulbs running the same firmware and yaml and one of them is no longer showing up in Home Assistant.

INFO Reading configuration /config/esphome/office-desk-lamp-left.yaml...
INFO Updating https://github.com/KaufHA/kauf-rgbww-bulbs@None
Failed config

external_components: [source /config/esphome/office-desk-lamp-left.yaml:16]
  - source: 
      type: git
      url: https://github.com/KaufHA/kauf-rgbww-bulbs
    refresh: 0s
    components: 
      - kauf_rgbww
      - light
      
      Could not find __init__.py file for component web_server. Please check the component is defined by this source (search path: /config/esphome/.esphome/external_components/69101fc3/components/web_server/__init__.py).
      - web_server
      - web_server_base

bkaufx, 2022-04-26:
It's weird that ESPHome would throw that error without first updating the 'common' github repo that that component comes from.  Does your office-desk-lamp-left.yaml file refer to https://github.com/KaufHA/kauf-rgbww-bulbs/blob/main/kauf-bulb.yaml as a package? (would use yaml as follows)

```
packages:
  kauf.rgbww: github://KaufHA/kauf-rgbww-bulbs/kauf-bulb.yaml
```

What are the contents of the /config/esphome/.esphome/external_components/69101fc3/components/web_server/ folder?

Try deleting folders /config/esphome/.esphome/external_components/ and /config/esphome/.esphome/packages/ again.  Restart ESPHome add-on as well.

Exit2Studios, 2022-04-27:
[image]

I see no mention of packages in the yaml.

bkaufx, 2022-04-27:
Sorry, that image is of an older yaml version.  I would recommend that you grab the newest [yaml ](https://github.com/KaufHA/kauf-rgbww-bulbs/blob/main/kauf-bulb.yaml)and more preferable would be to incorporate the github version as a package.

This is really all you need in `office-desk-lamp-left.yaml`:

```
substitutions:
  name: office-desk-lamp-left
  friendly_name: Office Desk Lamp Left

packages:
  kauf.rgbww: github://KaufHA/kauf-rgbww-bulbs/kauf-bulb.yaml

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

For the light that is offline and not showing up, does the light turn on at all or does it appear to be completely dead?  Have you checked for its fallback wifi hotspot?  Try powering down and back up the bulb a couple of times and then looking for the hotspot.

Exit2Studios, 2022-04-27:
The bulb turns on, but doesn’t connect to my router and I don’t see the hotspot. Is my only other option to flash it physically?

KaufHA, 2022-04-27:
Yea, that's probably what you're going to have to do.  If you can't get it, send your address to me at brian@kaufha.com and I can send a replacement. 

Another thing would be to just double check your router / dhcp server to make sure it's not to blame by either being full or blocking new connections for whatever reason.

bkaufx, 2022-06-01:
Make sure you are on the latest ESPHome version (2022.5.0 or 2022.5.1).  Also try deleting the .esphome/packages folder from within your ESPHome config directory.  Restart ESPHome addon.

aznnomness, 2022-06-01:
Home Assistant is reporting I am running the most recent version of ESPHome Add-on at version 2022.3.1. I will try deleting the .esphome/package directory.

bkaufx, 2022-06-01:
Make sure you backup all your esphome device yaml files first.

bkaufx, 2022-06-01:
This is the repo you need to add in the add-ons settings menu: https://github.com/esphome/home-assistant-addon

aznnomness, 2022-06-01:
Am I going to need to back up the `kauf-bulb-aux.h` and `kauf-bulb-main..h` files?

bkaufx, 2022-06-01:
No those files are not used anymore.   You don't need them anymore at all.  You only need these lines in your device yaml file and everything else is taken care of:

```
packages:
  kauf.rgbww: github://KaufHA/kauf-rgbww-bulbs/kauf-bulb.yaml
```

aznnomness, 2022-06-01:
Looks like that resolved my issues, thank you!

closed 2022-11-30 by KaufHA