KaufHA/kauf-rgbww-bulbs #12: Constant Unknown Reboots

marc-gist, 2022-06-08:
I have the light to turn on after boot...  well "Kauf Bulb Power On State" set to default bright white.  I want this feature, however, its turning on "by its self" constantly (notice on my wifi controller that the bulb has reconnected to wifi, signal is at 99% -39db, so not a wifi issue).

I'm on the latest: Firmware version 1.84(u) made using [ESPHome](https://esphome.io/) version 2022.5.0.

KaufHA, 2022-06-08:
That feature is working fine for me, so it sounds like maybe a hardware issue.  Were you able to confirm based on uptime or whatever that it is rebooting?  The only other thing I can think would do that would be potentially losing wifi, but doesn't sound like that is an issue for you.  If you email me brian@kaufha.com I can send out a replacement.

KaufHA, 2022-06-08:
Does it also reboot with other power on state options?

marc-gist, 2022-06-08:
> Does it also reboot with other power on state options?

Thanks for the quick relpies... not sure its hardware yet, trying to ensure its not some other issue.  I have changed the power on state to last.  Likely fix my "issue", not quite what I would like, but whatever :)
I'll keep playing with it. If it keeps happening I'll put Tasmota on it to rule out hardware as I'm more familiar with that and it provides a last reboot reason on its web interface and in mqtt logs.

marc-gist, 2022-06-08:
> 
Thanks!
about every 10min, so with it set to turn on after power loss (well, upon reboot), its annoying. otherwise working fine.  2 other bulbs so far appear to be ok. its a BR30 bulb, others are regular, not sure that matters, I'm now testing a second BR30 bulb to see if it does it as well.

marc-gist, 2022-06-08:
Found the issue. because I'm just testing the bulbs via the web gui
`22:14:10 | [E] | [api:127] | No client connected to API. Rebooting...`
I noticed this finally by leaving the web page open.
So its just an esphome issue.

bkaufx, 2022-06-08:
Oh yea, I should have thought of that.  Is the bulb in Home Assistant?  If you turn on the No HASS switch it will stop that from happening either way.  If it is in Home Assistant, I would remove and re-add.

BryanHaven, 2022-06-11:
I'm seeing a similar issue - I have several ceiling fixtures with three BLF10 bulbs each. 

All are flashed with the latest Tasmota Version 11.1.0(tasmota)-2_7_4_9(2022-04-13T06:40:42). 

 They are being managed with Homekit and Homebridge 1.4.1 with the homebridge-smartthings v2.3.8 plugin - on the Smartthings side using the Smartthings Tasmota (connect) & Homebridge v2. 

One bulb is frequently rebooting (every couple of minutes) but the others are not - I've reset the firmware back to the latest stock firmware and the behavior persists. At this point I'm thinking it's hardware but wanted to check in and see if there's something I'm missing before going that route.

KaufHA, 2022-06-11:
Probably hardware.  Send me an email.  For tasmota, make sure you use RGBWWTable to turn down the max power a bit to protect the power supply.  RGBWWTable 204,204,122,153,153 was one suggestion.  You could probably go a bit higher, but all 255's has a good chance of wearing out the power supply.

BryanHaven, 2022-06-11:
Will do - normally using the 4255k color and 70% - 80 % on the dimmer function. This one has only been in service for a few days.

Thanks,

closed 2022-06-21 by marc-gist