KaufHA/kauf-rgbww-bulbs #13: Effects

marc-gist, 2022-06-21:
Could you please show me how to add effects that can run on the bulb. I.e. transition between a list of colors?

Thanks!

KaufHA, 2022-06-21:
If you want to do [normal ESPHome light effects](https://esphome.io/components/light/index.html#light-effects) then you'll need to download the [kauf-bulb.yaml file](https://github.com/KaufHA/kauf-rgbww-bulbs/blob/main/kauf-bulb.yaml) to add them in, instead of including the yaml file as a package.  Basically copy/paste that yaml file into your yaml file and then edit it to change the name/friendly_name, wifi settings, and anything else you had in your yaml file to begin with.  You should be able to add effects under `id: kauf_light`

Something like a [lambda effect](https://esphome.io/components/light/index.html#lambda-effect) could be done while keeping the yaml included as a package by just adding your own [template switch](https://esphome.io/components/switch/template.html) to turn the effect on.  Have the switch start a [script](https://esphome.io/guides/automations.html#script-component) when turned on and stop the script when turned off.  The script can just call light.turn_on a bunch to turn to whatever color you want.  At the end of the script, call the script again.  You could also use a [button](https://esphome.io/components/button/template.html) instead of a switch.  When the button is pressed, check to see if the effect script is running.  If it is running, stop it.  If it is not running, start it.

The [factory yaml](https://github.com/KaufHA/kauf-rgbww-bulbs/blob/main/kauf-bulb-factory.yaml) file has an example of a script that changes the color in a loop.

natp13, 2022-09-19:
You would also need to add the following block back to light_state.cpp (see added code in --- block)

```c++
void LightState::loop() {
 
  // run wled / ddp functions if enabled
  if ( this->use_wled_ ) {wled_apply();}

  // if not enabled but UPD is configured, stop UDP and reset bulb values
  else if (udp_) {

    // stop listening on udp port
    ESP_LOGD("KAUF WLED", "Stopping UDP listening");
    udp_->stop();
    udp_.reset();

    // return bulb to home assistant set values instead of previous wled value
    this->current_values = this->remote_values;
    this->next_write_ = true;
   }

  // -----------------------------------------------
  // TODO: This is the code I had to add to make the candle effect play!
  // Apply effect (if any)
  auto *effect = this->get_active_effect_();
  if (effect != nullptr) {
    effect->apply();
  }
  // -----------------------------------------------

  // Apply transformer (if any)
  if (this->transformer_ != nullptr) {
    auto values = this->transformer_->apply();
    this->transformer_active = true;
    if (values.has_value()) {
      this->current_values = *values;
      this->output_->update_state(this);
      this->next_write_ = true;
    }
```

natp13, 2022-11-05:
@KaufHA added the effect->apply() back, so that's not necessary anymore, thanks!

techoguy, 2023-10-19:
To add effects, you can use the **!extend** feature in your yml file in ESPHome. below is an example of effects I added in my yml file. Once you add them, you can install it OTA to the light. Then use the light_on service call to activate the effect in HA. 

refer to the link to create your own effects or adjust existing effects: [https://esphome.io/components/light/#light-effects](https://esphome.io/components/light/#light-effects)

```
light:
  - id: !extend  kauf_light
    #https://esphome.io/components/light/#light-effects
    effects:
      - flicker:
          alpha: 94%
          intensity: 12%
      - strobe:
      - pulse:
      - random:
```

TikiBill, 2024-01-17:
@KaufHA Do you have any thoughts/plans on adding standard effects beyond WLED/DDP? I have what I consider a pretty nice candle effect as a lambda.