KaufHA/kauf-rgbww-bulbs #15: Bulb firmware fails unless paired to a network

jonofmac, 2022-07-14:
Hello,
This is building onto my other issue (not being able to connect to an SSID that's not broadcasted), so I figured since the firmware was out of date, I'd update it first and then see if it would connect.

Turns out trying to flash 1.85 onto these bulbs out of the box (connected to their own hotspot) fails. It acts like it uploads, and then chrome states the connection was reset by the bulb. 

Once I pair the bulb to a wifi network, then attempting to flash the newer firmware appears to work. Very interesting!

KaufHA, 2022-07-14:
Sorry about this.  One other person told me the same thing, but when I've tried it has always worked.  I'll take another look.

rake8288, 2022-10-12:
I am also seeing this occur with my bulbs. 

I'd be more than happy to provide any details needed to investigate.

bkaufx, 2022-10-12:
Interesting.  Are you able to update via the web interface once connected to wifi, but not via the wifi ap the bulb puts out?  I could potentially send a firmware with debug code to try to figure out the problem.

jonofmac, 2022-10-14:
Yes. That is the process I had to do:
I initially try to update firmware to latest with out of box bulbs. It doesn't work.
1) Pair to WiFi
2) Update using web interface

Update works.