KaufHA/kauf-rgbww-bulbs #16: BRF30 bulb run hot?

jonofmac, 2022-08-11:
Hello,
One of my flood light bulbs have failed. I suspect the switcher failed due to heat. It won't turn on with a power toggle even though it's set to power on. It won't connect to wifi and is not making an AP. I suspect the switcher failed, perhaps due to heat. I noticed while configuring the bulbs that they get quite warm. I purchased on Amazon and set up a replacement.

KaufHA, 2022-08-12:
They run about as hot as other LED bulbs I think.  I've noticed that the output capacitor on the power supply can go bad for some people, so I'm going to try to start using one that can withstand the temps better.  Did you get a replacement, or do you still need one?

jonofmac, 2022-08-12:
Amazon has a replacement on the way. I have to return the broken one, so
I'm not sure if it goes to you or not, otherwise I'd be happy to open and
check the cap out (I'm an EE) for you.

KaufHA, 2022-08-13:
It's pretty tough to get to the power supply on the BR30's so probably not worth you messing with it.  I think it will get back to me eventually.

jonofmac, 2022-10-14:
Just bumping, I've had 2 more bulbs die, assumingly due to heat. One of them I just pulled out of the box (I have 6 spares on hand still new in box). It's been out of the box for approximately 2 hours and now does not function. Putting a new light in and beginning to repair now. We'll see if the other bulb in this box sees the same fate.

I'd be willing to help look into this one more.

bkaufx, 2022-10-15:
Geez sorry man.  If you email me your address I can send you some more.  Haven't had anyone else with this many problems.  It tends to be the power supply that people have trouble with (it seems), which is pretty impossible to get to on the BR30's.  You can check the 3.3v pad on the LED PCB and see if there is power there.

jonofmac, 2022-10-16:
Haha. Im not sure what it is. The tattoo that have died have been in lamps
with open tops (on surge suppressors too, but i don't have power issues and
none of my ceiling light ones have had issues except for 1 flood light that
died). The ones in ceiling fans have been fine surprisingly so I'm not sure
the root cause. I imagine i have to cut the diffuser off to get to the PCB?
Quick look shows the diffuser probably isn't meant to come off I've
installed.

I really do like the bulbs, just not sure what is killing them.

My address is 1813 Lincolnshire Dr. Bedford, TX 76021

JoshHawley, 2022-11-24:
I also had one of these BR30 bulbs die. I got it in July, and it seems completely dead. The wife says she heard an audible pop when it went dark. 
 
 What is the right way to contact support about these? If there is a different way / place to contact you about these please let me know.

bkaufx, 2022-11-24:
Send email to brian@kaufha.com.

gingerbreadassassin, 2023-03-01:
I've had 60-80% of my BR30 bulbs die. I subsequently found some pretty scary DIY electrical work in my new-to-me house from 1996, including but not limited to:

1. Shared neutrals
2. Orphaned cables
3. Bonded neutrals and grounds in the sub-panel for the house (main panel is out on the utility pole)
4. Loose hots in the aforementioned sub-panel

When they die, they tend to dimly flicker at somewhere between 1-2Hz when power is applied, and slowly taper off in brightness and frequency when power is removed. The ones that have survived work great, though.