KaufHA/kauf-rgbww-bulbs #18: Suggestion: ESP32 for WPA3 + better antenna

NeoMatrixJR, 2022-09-19:
Sorry, not sure where else to post this.  I love my bulbs so far other than I have issues keeping them connected to my WiFi APs.  Any chance for a better antenna design?  Also, would like to see an upgrade to ESP32 to support WPA3 standard.  You could use an ESP32-S2 if you wanted to save a bit and avoid the extra Bluetooth component, extra cores, etc...  But I kinda wish more of my devices had ESP32s with BLE for tracking....

KaufHA, 2022-09-22:
I'm trying to get devices with ESP32.  It's tough when I'm pretty much just private labelling existing products so far.  It sounds like the factory can replace ESP8266 modules with ESP32 modules in some cases, but the bulbs aren't really as modular as some other products.  I'll ask them about it though.

Sorry you aren't keeping a consistent Wi-Fi connection.  I've never had any issues and haven't had any complaints really.  Is the signal weak?  You can add a wifi signal strength sensor pretty easily with the code here: https://esphome.io/components/sensor/wifi_signal.html

Are you using this with Home Assistant?

NeoMatrixJR, 2024-02-15:
Turns out I may have had an electrical connection issue with one of my bulbs...and it happened to be the furthest one from my AP so it appeared to be a connectivity issue.  Unfortunately I found damage to the bulb and socket....not sure what the cause was but it looks like there was probably some gap sparking that damaged both.  I've replaced the bulb and fixtures (they needed an update anyway) and have seen a marked improvement.  Thanks for helping to provide the community with open firmware devices.  It's great to see in a Tuya world!

closed 2024-02-15 by NeoMatrixJR