KaufHA/kauf-rgbww-bulbs #19: Color temperature incorrect with low main brightness and warm_rgb on

natp13, 2022-09-19:
### Summary
When setting the brightness of the main bulb to low values (<20%) with warm_rgb turned on, the actual color is incorrect, and sometimes even flickers (17% is especially bad). This wouldn't really be an issue except when using these bulbs with the flicker effect to simulate a candle, the drastic change in color temperature whenever the brightness drops too low is not good. And there's not a great way to get the correct color without using the warm_rbg.

### More details:
- color temperature on main bulb set to 350mireds
- warm_rbg bulb turned on and set to reddish orange (white: 40%, red: 100%, green: 73%, blue: 0%)
- Version 1.845 of kauf_bulb.yaml (with a few custom additions). Full yaml copied below
- Added the effect->apply() code to light_state.cpp so effects will work.
- I'm pretty sure this color issue also happens without any of this custom code, but I haven't had a chance to try lately.
- I started digging into the color math that happens, but that was looking like a deeper dive than I had the time for.

Full yaml:
```yaml
# esphome upload desk.yaml --device <orignal ip address>

# https://esphome.io/guides/configuration-types.html#substitutions
substitutions:

  # substitutions can be changed here if you are using this file directly in the ESPHome dashboard.  The better approach is
  # to incorporate this file as a package using the following lines, and then overwrite these substitutions in your local
  # yaml file by redefining them.
  #
  # packages:
  #   kauf.rgbww: github://KaufHA/kauf-rgbww-bulbs/kauf-bulb.yaml
  #

  name: kauf-bulb              # **** CHANGE DEVICE NAME TO SOMETHING UNIQUE PER DEVICE.  RENAME YAML FILE TO SAME NAME.    ****
                               # **** USE DASHES (-) INSTEAD OF SPACES OR UNDERSCORE (_).  USE ONLY LOWER CASE LETTERS.     ****

  friendly_name: Kauf Bulb     # **** CHANGE FRIENDLY NAME TO SOMETHING UNIQUE PER DEVICE ****

  disable_entities: "true"     # set to "false" to have all entities show up in Home Assistant automatically

  # https://esphome.io/components/light/index.html#base-light-configuration
  light_restore_mode: RESTORE_DEFAULT_OFF   # change power-on action for main light entity - NAT CHANGED
  default_power_on_state: "Restore Power Off State"

  wifi_ap_timeout: 2min    # default to 2 minute timeout for yaml file as package.

  # https://esphome.io/components/esphome.html#esphome-creators-project
  project_name: Kauf.RGBWW
  project_ver_num: "1.845"
  project_ver_let: y

  sub_on_turn_on:  script_do_nothing
  sub_on_turn_off: script_do_nothing

# https://esphome.io/components/esp8266.html
esp8266:
  board: esp01_1m
  restore_from_flash: true
  start_free: 76
  global_addr: global_forced_addr

# https://esphome.io/components/external_components.html
external_components:
  - source:
      type: git
      url: https://github.com/KaufHA/common
    refresh: 0s
  - source:
      type: git
      url: https://github.com/KaufHA/kauf-rgbww-bulbs
    refresh: 0s
  - source:
      type: local
      path: copied_components

# https://esphome.io/guides/automations.html#global-variables
globals:
[... 397 lines cut ...]
          kauf_light->set_restore_mode(LIGHT_RESTORE_AND_ON);
        else if ( id(select_boot_state).state == "Always On - Bright White")
          kauf_light->set_restore_mode(LIGHT_ALWAYS_ON);
        else if ( id(select_boot_state).state == "Always Off")
          kauf_light->set_restore_mode(LIGHT_RESTORE_AND_OFF);
        else
          kauf_light->set_restore_mode(LIGHT_$light_restore_mode);

  - id: script_do_nothing
    then:
      lambda: return;

# Current reserved flash memory:
# 00-25: Wi-Fi Credentials

# 28-39: Warm RGB
# 40-51: Cold RGB
# 52-63: Main Light
# 64-65: Power On State select
# 66-67: Effect select

# 70-71: No HASS switch
# 72-73: Default Fade Length
# 74-75: Max Power
```

bkaufx, 2022-09-22:
Thanks adding so many details.  I'll check it out this weekend.

bkaufx, 2022-09-24:
Did the flicker effect work for you with the main light set to 350 mireds?

In addition to adding the effect->apply() code back in I had to add in a line to base_light_effect.h to copy the color temperature from remote to out (Line 222:  `out.set_color_temperature(remote.get_color_temperature());`).  Otherwise it would always flicker with the cold white LEDs and never warm white.  Meaning, turning the Candle effect on would change the light's display from warm white to cold white even though it still showed warm white in Home Assistant.

BTW I'll add the effect->apply() code back in so you don't need your own custom components.  Still looking at the flickering issue, I have a couple of ideas.

bkaufx, 2022-09-25:
Try turning down the frequency on the PWM outputs.  I think I'm doing too much on the ESP8266 to be able to also have 5 channels of software PWM running at 1000Hz, so it causes inaccuracies in timing.  I used the following and it seemed to improve.  I used 400Hz for the white channels and 200Hz for the colors.  I'll probably play around with it more and find new values for the release firmware.

Lowering the PWM rate also increases resolution at low brightness so should help with accuracy that way as well.

```
# PWM outputs for each LED channel
# https://esphome.io/components/output/esp8266_pwm.html
output:
  - platform: esp8266_pwm
    pin: GPIO4
    frequency: 200 Hz
    id: pwm_red
    max_power: .85
  - platform: esp8266_pwm
    pin: GPIO12
    frequency: 200 Hz
    id: pwm_green
    max_power: .85
  - platform: esp8266_pwm
    pin: GPIO14
    frequency: 200 Hz
    id: pwm_blue
    max_power: .85
  - platform: esp8266_pwm
    pin: GPIO5
    frequency: 400 Hz
    id: pwm_cw
    max_power: .85
  - platform: esp8266_pwm
    pin: GPIO13
    frequency: 400 Hz
    id: pwm_ww
    max_power: .85
```

natp13, 2022-11-05:
Finally got a chance to come back to this. I copied over the latest version (1.863) from kauf_bulb.yaml and this seems to have fixed it! Thanks for adding back the effects piece too!

natp13, 2022-11-05:
Ah spoke too soon. It's definitely better, but still flickers around 10% brightness and below.

bkaufx, 2022-11-05:
Hmmm. Can you reply with complete exact settings of main light and cold/warm rgb lights?  color and brightness.  If you go to the states tab under developer tools it will show you the exact rgb values etc.

SpencerDub, 2023-02-14:
I'm having a similar issue with color temperatures: at their warmest settings, the bulbs are still relatively cool. Looking at the States tab, I see the bulbs have `min_color_temp_kelvin: 2857`, where for other of my bulbs, that's `2000` (allowing for much warmer light like I'm looking for).

Can I change that value to enable warmer light? If so, how?

natp13, 2023-02-14:
> I'm having a similar issue with color temperatures: at their warmest settings, the bulbs are still relatively cool. Looking at the States tab, I see the bulbs have `min_color_temp_kelvin: 2857`, where for other of my bulbs, that's `2000` (allowing for much warmer light like I'm looking for).

Not necessarily related to the main issue, but the color temperature of the bulbs definitely shifts towards cool as the bulbs warm up from being on. Apparently this is common with LEDs, but I wonder if the color temperature of these bulbs is the steady state, or the initial on temp...

natp13, 2023-09-19:
> Hmmm. Can you reply with complete exact settings of main light and cold/warm rgb lights? color and brightness. If you go to the states tab under developer tools it will show you the exact rgb values etc.

**main_light:**
min_mireds: 150
max_mireds: 350
effect_list: None, Candle
supported_color_modes: color_temp, rgb
friendly_name: Kauf Bulb
supported_features: 44
color_mode: color_temp
brightness: 26
color_temp: 350
hs_color: 28.077, 60.925
rgb_color: 255, 172, 99
xy_color: 0.51, 0.386
effect: None

**cold rgb off**

**warm rgb**
supported_color_modes: rgbw
icon: mdi:store-cog
friendly_name: Kauf Bulb Warm RGB
supported_features: 40
color_mode: rgbw
brightness: 255
hs_color: 44.409, 49.804
rgb_color: 255, 222, 128
rgbw_color: 255, 189, 0, 255
xy_color: 0.437, 0.421