KaufHA/kauf-rgbww-bulbs #21: Can only add through integration

safadig, 2022-10-08:
wifi bulbs are on different subnet with no mDNS. I can add each bulb manually as Integration.  Ping enabled. Anyway I can manually add to ESPHome dashboard after they are added? For now, the Dashboard is empty.

bkaufx, 2022-10-08:
I'd recommend removing from home assistant, adding to the esphome dashboard and reprogramming each device, then readding to home assistant.

Create a yaml file in the esphome config directory for each bulb. Copy in this yaml to each file:

```
substitutions:
  name: bedroom-lamp
  friendly_name: Bedroom Lamp

packages:
  kauf.rgbww: github://KaufHA/kauf-rgbww-bulbs/kauf-bulb.yaml

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

Rename each as desired. Add either a use_address or a manual_ip under each wifi section to point to the respective device per these directions: https://esphome.io/components/wifi.html

Then hit install in esphome and do it wirelessly.

safadig, 2022-10-08:
> I'd recommend removing from home assistant, adding to the esphome dashboard and reprogramming each device, then readding to home assistant.
>
> Create a yaml file in the esphome config directory for [...]

Tried with 2 devices and ESPHome Dashboard gave option to update so I did. Unfortunately both devices now are not usable as I can see the IP Address but:
```
INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.xxx.175 using esphome API
WARNING Can't connect to ESPHome API for 192.168.xxx.175: Timeout while connecting to ('192.168.xx.175', 6053)
INFO Trying to reconnect to 192.168.xxx.175 in the background
```
Both Bulbs turn off and will not come back on! I cannot access the IP address page either