KaufHA/kauf-rgbww-bulbs #22: New BR30 Bulbs asking for wifi password

safadig, 2022-10-08:
Just got new 4 of BR30s and the Kauf Bulb Hotspot prompts for a password. 
Tried asdfasdfasdfasdf but did not work. Any Suggestions?

bkaufx, 2022-10-10:
There's no reason that should be happening.  I'd try a different device to join the network.  You could also try setting up a wifi network with ssid initial_ap and password asdfasdfasdfasdf to see if the bulbs connect to that.

safadig, 2022-10-10:
did all of that. he kauf bulb hotspot is locked with a password. I can't imagine it is just me running into this

bkaufx, 2022-10-11:
Never heard of this before.  The only way I can image this happening is if a customer added a password and returned the bulbs, then Amazon re-sold them without sending them back to me.  E-mail me your address and I'll send you another one to test. brian@kaufha.com.  If the new one works fine I'll send 3 more.  

What is the exact name of the hotspot being put up by the bulbs?  Kauf Bulb Hotspot?

A couple other things to try:
- make a network with ssid initial_ap2 instead of initial_ap, same password asdfasdfasdfasdf
- take the bulbs off-site and try to set up somewhere else.
- you could always manually reflash with wires: https://github.com/KaufHA/kauf-rgbww-bulbs/issues/1

safadig, 2022-10-15:
@bkaufx , Thanks for the offer. I just ended up following the Jig Flashing procedure. Flashed them with 1.85 and now everything is fine. Must have been as you said, someone returned them after changes

closed 2022-11-14 by KaufHA

Jord4n, 2023-09-02:
I have the same issues, except I think I locked myself out of the bulb with an AP that I lost the password to due to a Home Assistant reset.  

How do you pop off the bulb from the base?  The bulb is broadcasting an AP, so I know it's not dead.  However, it will not turn on at all after trying the power on/off 5 times under 10 seconds.

bkaufx, 2023-09-03:
You might try setting up a wifi network with ssid initial_ap and password asdfasdfasdfasdf.  The bulb should try to connect to that, then maybe you can ota from there.  To pop the top off just get something grippy and twist.

Jord4n, 2023-09-04:
Hi @bkaufx thanks for the reply!  I tried both initial_ap and inital_ap2, no luck.  Even with a fresh HA & Esp install.

I twisted the bulb from the base, and it rotates free now.  How do I actually get the bulb off, though?  Seems on tight and when I rotate I keep getting the "clicking".  Am I supposed to pry it off?  Pull hard?

bkaufx, 2023-09-04:
Sorry I was thinking BR-30.  Those come off super easy once you twist.  ON the A21/A19 bulbs, you do have to kind of pull pretty hard.  What I usually do is get two pieces of something grippy, one for each part (I use some cut up shelf liner, but think like things you would use to open a jar).  Using those to grip the two parts, I press down the bulb on my leg like a kid trying to break a stick in two.

Jord4n, 2023-09-05:
All good.  That was scary!  I finally got it off with a plastic spudger, but this whole tome I thought it was a traditional GLASS bulb and not plastic! I was sweating so hard with safety glasses on! LOL!

I downloaded [this jig](https://www.thingiverse.com/thing:5269303) to try and re-flash.  

Just to be clear (this is an A21 RGBWW bub) what firmware should I re-flash them with?

bkaufx, 2023-09-05:
https://github.com/KaufHA/kauf-rgbww-bulbs/releases/download/v1.87/kauf-bulb-v1.87-update.bin

you'll need the .bin file, not the .bin.gz, to flash via wire.