KaufHA/kauf-rgbww-bulbs #24: Unable to update any lights

GinAndBacon, 2022-11-17:
First time I have ever had an issue but during the latest update I get

Failed config

select: [source /config/esphome/.esphome/packages/0cf47246/kauf-bulb.yaml:223]
  
  Component select cannot be loaded via YAML (no CONFIG_SCHEMA).

and then

number: [source /config/esphome/.esphome/packages/0cf47246/kauf-bulb.yaml:301]
  
  Component number cannot be loaded via YAML (no CONFIG_SCHEMA).

Attached are the full logs. Each work with zero issues through HA or going directly to the IP. I also have the Kauf power switch and get the same error but with a different line number and only one line number instead of two. Really don't feel like reinstalling all four lights. Did not create an issue for the power switch yet as it looks like both of the files that it references (kauf-bulb.yaml) was updated 18 hours ago. Below is the config for one of the four lights. All have the same config besides name/friendly name. I also have four lights from another vendor and they all updated with zero issues today.

substitutions:
  name: color-light-four
  friendly_name: color light four
packages:
  kauf.rgbww: github://KaufHA/kauf-rgbww-bulbs/kauf-bulb.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  use_address: 192.168.0.23

[kauf_light_error_log.txt](https://github.com/KaufHA/kauf-rgbww-bulbs/files/10032139/kauf_light_error_log.txt)

bkaufx, 2022-11-17:
Try cleaning build files.  If that doesn't work, delete the .esphome/packages and .esphome/external_components subdirectories within the ESPHome config directory.

I removed the custom number and select components since they weren't needed.  For some reason you are stuck looking for those components in the custom folder and not using the stock components.  I think the above steps will fix it.  Let me know if not.

GinAndBacon, 2022-11-17:
Cleaning the build files resulted in the same error but deleting the .esphome/packages and .esphome/external_components subdirectories fixed the issue and I just updated on light with zero issues. You can close (unless I have to close) this issue. Many thanks to your quick and timely response and way to resolve.

KaufHA, 2022-11-30:
I figured out a workaround for this so shouldn't be a problem for anyone else.  Basically, always referring to a tagged version of the common github repo so that the ESPHome dashboard will place the updated repo into a new folder without empty subfolders.

closed 2022-11-30 by KaufHA