KaufHA/kauf-rgbww-bulbs #26: BR30 Bulbs disconnecting from Wifi

applehomekit1, 2022-12-04:
I have a full Ubiquiti Unifi network setup running my wifi. I have had 3 bulbs so far in the last 2 months (I just purchased them in May 2022) consistently disconnecting from wifi and becoming unlinked to home assistant. I have had to power them off and on several times to get them to reconnect to wifi. None of my wifi settings have changed. I have had 1 bulb just yesterday go offline and will not turn on or reset. I have 20 of these bulbs. All have the latest esphome installed via home assistant. Any ideas, I don't want to waste money on anymore if this is going to continue.

applehomekit1, 2022-12-06:
Some of them will stop responding to home assistant or my smart switches controlling them, and when I inspect them, notice they are not connected to my wifi anymore. I then found you have to unscrew light, then screw back in (sometimes several times) for it to reconnect to the wifi. I have a wifi access point in this room with 12 of these lights. I have 1 main one doing this, but have 3 that have done it a few times in last month. I think by default fast_connect is off, I don't have it turned on. I just ordered a new replacement bulb for the one that died from Amazon.

maverickccie, 2024-08-15:
I have 4 of these bulbs and two of them regular disconnect from my Ubiquiti AP's then take 5 to 10 mins to reconnect. But I have 10 of the a21 bulbs that never have an issue.  Has anyone found a fix for this? I notice that when the two BR30s drop when they reconnect, I see this in the logs:
Could not allocate memory for JSON document! Overflowed largest free heap Block

applehomekit1, 2024-08-15:
I never found a fix. They replaced a few bulbs for me. I ended up replacing all with zigbee bulbs and fixed my problems.

maverickccie, 2024-09-16:
I think I found a fix. I Isolated them to a single AP. So instead of one IOT SSID I now have one for each AP. Did this 5 days ago and so far, not a single ESPhome device including my Kauf bulb has disconnected.  I am starting to think that ESP based devices cannot move from AP to AP smoothly on Ubiquiti. Just another reason I will be replacing Ubiquiti with Cisco come January.