KaufHA/kauf-rgbww-bulbs #28: RGBWW Bulb with xLights DDP weird brightness control problems

PKCubed, 2022-12-07:
I got some bulbs for my Christmas light show and connected them up to my HA network to test. They seem to work with xLights but I have weird issues with color. I can play an effect where the bulb should fade from black to red, then to black again, or for any color or color combination, however, the bulb just stays the color without changing brightness. I know this is not wifi slowness or anything because when I have it on a more "exciting" effect where colors are always changing, I still cannot fade to black. It will just jump to black when the effect reaches full blackness. Any help? Thanks!

bkaufx, 2022-12-07:
Try turning the lights off in Home Assistant.  I made it such that the DDP packets will be scaled to the home assistant brightness if the home assistant light is on, but use the DDP packet raw data if the home assistant light is off.  I might change this as it is admittedly kind of unintuitive.  But for now, you need to turn the light off in home assistant.

PKCubed, 2022-12-08:
Thank you! I figured this out eventually. If possible, I recommend you change it so that the brightness slider in HA changes the values of the full intensities instead of the 0 values to 100% (I don't know a better way to word that).

PKCubed, 2023-02-25:
I'm having this proble

PKCubed, 2023-02-25:
I'm having this problem again as I moved my lights to a different network that doesn't have a home assistant server. Is there a way to turn the ha brightness down without home assistant, just through the web ui? Or do I have to set up a computer running HA server just to fix this problem? Other than this problem my lights seem to be working great.

bkaufx, 2023-02-27:
the web server should have a slider if you type the ip address into a web browser.  You can also automate it through the http api with, e.g., the following curl command, where the brightness should be an integer from 0-255

curl "http://192.168.86.201/light/kauf_bulb/turn_on?brightness=150" -X POST

PKCubed, 2023-02-27:
Thank you! I don't see the brightness slider on the web interface, but I
will try the curl command.

closed 2023-05-15 by PKCubed