KaufHA/kauf-rgbww-bulbs #29: RGBWW light is not creating WiFi AP

SpencerDub, 2023-02-14:
I just plugged a brand-new RGBWW bulb in and powered it on, and it has not created an access point. What's more, I've tried rebooting the bulb five times (as described [here](https://github.com/KaufHA/kauf-rgbww-bulbs#clearing-wi-fi-credentials-to-get-wi-fi-ap-back)) _and_ power-cycling the bulb before it can connect to WiFi (as described [here](https://github.com/KaufHA/kauf-rgbww-bulbs#clearing-wi-fi-credentials-to-get-wi-fi-ap-back)), and neither has worked. The bulb is powered on and displaying cold white light, but it is not broadcasting any WiFi access point that I can find.

SpencerDub, 2023-02-14:
Additional context: I've tried two other bulbs from the same shipment, and both established access points without any issue.

bkaufx, 2023-02-14:
Could have been a return that Amazon shipped back out to you automatically, flashed by someone else with who knows what.  You could try putting up a wifi network with the default credentials initial_ap/asdfasdfasdfasdf or soldering wires to flash.  If you email me your address I'll send a replacement.  brian@kaufha.com

SpencerDub, 2023-02-14:
Thanks for the quick response, Brian! I'll send you an email. 

> You could try putting up a wifi network with the default credentials initial_ap/asdfasdfasdfasdf

Those credentials literally, or do you mean the credentials the bulb should have had? And once I do that, what next?

I haven't yet practiced on my soldering iron, so I'll have to pass on that option for now.

bkaufx, 2023-02-14:
Yea those credentials specifically are the default that the bulbs try to connect to until the user enters their own credentials.  If you put up a network like that, the bulb may connect to it.  Then you just have to figure out its IP address and type it in a web browser.  From there you can flash the original firmware back or a custom firmware.  If that doesn't work I'll send a replacement.

closed 2023-04-24 by SpencerDub