KaufHA/kauf-rgbww-bulbs #30: V1.875 wont connect to wifi. Keeps reverting to captive portal AP.

liamrpowell, 2023-02-15:


bkaufx, 2023-02-16:
Yea /reset won't work on the captive portal.  I've thought about adding it.  You can connect to the bulb's AP and enter credentials but they don't get saved?  Is there a table at the bottom with information?  What does it say?

bkaufx, 2023-02-16:
I should mention I updated some of my bulbs with esphome 2023.2.0 and our 1.875 yaml and they reconnected back to the same wifi network fine.

liamrpowell, 2023-02-16:
Correct, they just

> Yea /reset won't work on the captive portal. I've thought about adding it. You can connect to the bulb's AP and enter credentials but they don't get saved? Is there a table at the bottom with information? What does it say?

Correct, I add the SSID and password. But they never reboot and try to connect just sit there then the bulb resets and connects back to AP portal

liamrpowell, 2023-02-17:
Is there a way to totally factory reset the bulbs?

liamrpowell, 2023-02-17:
I also tried to upload a bin file using the ap, but as soon as I try it goes to a webpage stating invalid request

applehomekit1, 2023-02-24:
I am also having the issue with all my bulbs, I stopped updating them, and the ones i did update I had to put tasmota on them to work. I have updated my unifi APs and they still cannot see my network, they can see others. This only happened after this update. Any ideas?

applehomekit1, 2023-02-24:
For testing, I created a new wifi network with nothing but letters and numbers and no spaces, upon reconnecting to bulb, it did see the wifi, and was able to connect and control. The settings are identical to the previous network except no special characters.

bkaufx, 2023-10-26:
Can you try creating your own firmware file with your credentials built in?  You would do this from the ESPHome dashboard and do manual install.  Then gzip the bin file.  Then upload through the captive portal.  If the web interface for the captive portal isn't working to upload the firmware some people have had better luck with using the curl command.  

`curl -v -X POST -H "Content-Type: multipart/form-data" -F update=@firmware_filename.bin.gz  "http://192.168.4.1/update"`

If that doesn't work send me an email and I can send a replacement to see if it works better.

ThrashinVictim, 2023-10-31:
> Can you try creating your own firmware file with your credentials built in? You would do this from the ESPHome dashboard and do manual install. Then gzip the bin file. Then upload through the [...]

I think I have a bad bulb. I finally got it to take the firmware, but if I leave it on for 15-30 minutes it starts flashing again. But it doesn't lose wifi because I can ping it.