KaufHA/kauf-rgbww-bulbs #33: How to change max color temperature?

SpencerDub, 2023-04-25:
When using the warm RGB setting, I can set a Kauf bulb to output light that roughly matches a non-Kauf bulb set to a temperature of 2000K. However, the Kauf bulb's temperature range is capped at 2857K, so despite matching the color of 2000K, the attribute is 2857K.

How can I manually override the minimum temperature of the bulb, so its numeric range more accurately represents the light I'm able to produce using the warm RGB setting?

Thanks!

bkaufx, 2023-04-25:
On the normal ESPHome light you do it via cold_white_color_temperature and warm_white_color_temperature settings .  Seen here in the lite yaml: https://github.com/KaufHA/kauf-rgbww-bulbs/blob/main/kauf-bulb-lite.yaml#L159-L160.

I will look into making this over-rideable on the default yaml.

bkaufx, 2023-04-25:
Yea I think this should be easy enough.  I'll try to get to it in the next week or two.

SpencerDub, 2023-04-25:
Thanks, Brian! I appreciate it.

bkaufx, 2023-05-06:
Just added substitutions that can be overwritten to define min/max color temp.  Either mired or Kelvin can be used and you have to give the units with the value.

```
substitutions:
  sub_warm_white_temp: 350 mireds
  sub_cold_white_temp: 150 mireds
```