KaufHA/kauf-rgbww-bulbs #34: Bulbs reverting to AP, but AP won't stay connected

scorfman, 2023-05-07:
I have 8 or so of these bulbs running ESPHome. For some reason 3 of them will no longer connect to my network, and broadcast their AP. When I connect to the AP, the network immediately disappears (either right after seeing the GUI, or even before that at times) and reappears a few seconds later, like the bulb is resetting or something.

I've done the power-cycle wifi reset with no improvement. Any ideas?

robertsorensen, 2023-06-06:
I've had something similar happen twice.
The first time (and the details are a bit blurry by now) I tried several times to get my bulb back into home assistant without luck. The bulb was seen by my access point and had the right IP address but not until I restarted the access point did it reconnect to Home Assistant.
The 2nd time (the same bulb) the bulb did create its own network and I could log into it from my phone. However even when I entered my WiFi credentials it did not pick them. I removed power for an extended time (like hours); no change. I did try to reset my access point as the first time; no change. I did move the bulb to a different spot and still the same. 
I can't really say what happened but I went on a 3 week trip and when I got back I tried to again to enter my WiFi credentials on the bulb Web-server and it jumped right on and as of now works fine.
Properly not a lot of help but - you could try to book a 3-week trip to Asia and see what happens when you get back :-)

bkaufx, 2023-06-07:
Sorry I never responded to this the first time a month ago.  Any time the Wi-Fi captive portal isn't working right I always recommend setting up a wifi network with credentials initial_ap / asdfasdfasdfasdf.  The bulb should connect to that then you can flash a new firmware with credentials for your network burned into the bin file.