KaufHA/kauf-rgbww-bulbs #35: xLights DDP slight latency and inconsistancy.

PKCubed, 2023-05-15:
These bulbs are awesome. I set them up in my Christmas Light Show for next year, and they work pretty well except when using the default firmware and sending ddp to 3 bulbs (not using the weird thing where the first bulb hands the rest of the data to the next and the next...) I have slight latency and some inconsistency, lag, etc. I'm a perfectionist, and I need my light show to be timed so well you can't see any lag. I presume this latency is introduced because the default firmware seems to have DDP added as an afterthought, and is not optimized for it. Before these kauf bulbs I was using an esp8266 with wifi, WLED, and some WS2811 LEDs. It worked great. No performance issues at all, other than the occasional tiny lag spike every hour or so. Not a huge deal.

Am I correct in thinking that this latency is caused by the firmware?
**Is there any other firmware I can flash onto these bulbs that is built for DDP or e1.31?** I know WLED is not supposed to be flashed.

Thank you all!

bkaufx, 2023-05-15:
Tasmota has a DDP version you can try.  You can also try rolling your own ESPHome firmware with our minimal yaml and ddp component:

- https://github.com/KaufHA/kauf-rgbww-bulbs/blob/main/kauf-bulb-minimal.yaml
- https://github.com/KaufHA/common/tree/main/components#ddp

Was this something that worked well and got worse?  Or has there always been a lag?  I've never really noticed any lag but I don't use DDP a lot.

PKCubed, 2023-05-16:
Thanks for the information! As far as I can tell, the lag has been here. I just got these bulbs working with the show yesterday. Its only noticeable when you see the bulbs flashing at a slightly different time than the rest of the house. The rest of the house is all wired ethernet, and I haven't had any lag problems now that it is all ethernet. If it helps, I uploaded a short video of this behavior to YouTube, you can find it here:

https://youtu.be/QZkAc62xniI

In addition, here is a video of the same sequence but just the xLights preview (what the bulbs should be doing).

https://youtu.be/sOO4vmVqr5c

PKCubed, 2023-05-16:
How might I go about installing the Tasmota DDP firmware? I've never heart of Tasmota before.

bkaufx, 2023-05-16:
This dude created a binary with the functionality built in, so there is some info there.  https://cloudfree.shop/blog/2022/02/17/pair-tasmota-with-wled-for-lighting-effects/

1.  This is the bin.gz file you flash: http://tasmota.cloudfree.io/tasmota-ddp.bin.gz
2. It will put up a hotspot called Tasmota something so connect to it and enter your network credentials.
3. Browse to the IP address of the bulb.
4. Configuration -> Configure Other.  Paste in this template at the top and check activate:
  - `{"NAME":"Kauf Bulb", "GPIO":[0,0,0,0,416,419,0,0,417,420,418,0,0,0], "FLAG":0, "BASE":18, "CMND":"SO105 1|RGBWWTable 204,204,122,153,153"}` 
5. Click console and type in 'scheme 5'

Now it should be listening on DDP.  Try one bulb and see if it works better.

PKCubed, 2023-05-16:
Thank you! I'll try that out.

PKCubed, 2023-07-02:
Tasmota is working great! I think the latency and inconsistency problem was more due to my 2.4ghz network being full because of my neighbors. The issue now is that the latency is fine, but the "frame rate" is pretty low. Is there any way to increase this? It looks choppy.

bkaufx, 2023-07-02:
For Tasmota, I'm not sure.  You might try ESPHome again if you got the Wi-Fi cleared up a little bit.  You could also try our minimal yaml with the DDP component to see if it works any better than our stock firmware.  You might need to copy/paste the entire minimal yaml into your local config.

https://github.com/KaufHA/kauf-rgbww-bulbs/blob/main/kauf-bulb-minimal.yaml
https://github.com/KaufHA/common/tree/main/components#ddp

PKCubed, 2023-09-19:
I'm trying to flash the original firmware back on OTA. It says there is not enough space. How should I do this?