KaufHA/kauf-rgbww-bulbs #36: Configurable PWM Hz

csnyder616, 2023-05-25:
I use a couple of Kauf bulbs to light up my face for video calls. I've noticed that there's some flickering that I can't get rid of using webcam settings, and I'm wondering if it will be better if I increase the PWM frequency. Would it be possible to make the PWM frequencies configurable rather than hard-coded in kauf-bulb.yaml?

bkaufx, 2023-05-25:
Alright, check it out.  You might have to delete the .esphome/packages folder to get the ESPHome dashboard to see the update.

closed 2023-05-25 by csnyder616

Borogroves, 2024-12-09:
Could you point me in the direction of where the PWM frequency can be adjusted?

The bulbs have firmware 1.95, and I'm using ESPHome within HA.

bkaufx, 2024-12-10:
Add these substitutions in your yaml:

```
  sub_red_freq:   200 hz
  sub_green_freq: 200 hz
  sub_blue_freq:  200 hz
  sub_cw_freq:    400 hz
  sub_ww_freq:    400 hz
```

https://github.com/KaufHA/kauf-rgbww-bulbs/blob/main/kauf-bulb.yaml#L42-L46

Borogroves, 2024-12-13:
Thanks. I'm new to Home Assistant with ESPHome integration. So I don't have the full ESPHome dashboard. My questions then are more about how to split yaml configuration files and how to use packages.

In my main configuration.yaml, would I just include the kauf-bulb.yaml?

```
homeassistant:
  packages: !include kauf-bulb.yaml
```

And then in the kauf-bulb.yaml, this is the entire file contents (as no other substitutions are needed)

```
substitutions:
  sub_red_freq:   200 hz
  sub_green_freq: 200 hz
  sub_blue_freq:  200 hz
  sub_cw_freq:    400 hz
  sub_ww_freq:    400 hz
```

bkaufx, 2024-12-13:
No you need to install the ESPHome dashboard and import the bulb into the dashboard.  I think they call it "take control" now or something.

Then add those substitutions into the substitution section that should already exist in the bulb-specific yaml file automatically.

I'd also probably recommend adding a friendly_name substitution and then removing and re-adding the bulb in home assistant.