KaufHA/kauf-rgbww-bulbs #37: default on colour

arfletcher, 2023-06-19:
I have a number of Kauf A21 RGBWW Smart Bulbs running the latest esphome configs, all connected to Home Assistant. The bulbs work well - I can change the colour, etc..

Whenever I turn the bulbs on they are always set to orange (255 183 77). 

Can we set the bulb to keep the last colour setting or, at least, set a default colour (eg white)?

KAUF RGBWW Bulb by Kaufman Home Automation
Firmware version 1.91(y) made using ESPHome version 2023.5.5.

bkaufx, 2023-06-19:
What is the "Power On State" setting set to.  It should be working as you are wanting.  Maybe ESPHome changed something in a recent update.  I'll look into it.

arfletcher, 2023-06-19:
The "Power On State" is set to "Restore Power Off State".  I just tried all the options and whenever I turn the bulbs on they are set to orange. :-)

Thanks for checking. Happy to experiment.

bkaufx, 2023-06-19:
Do you have anything in the yaml besides what is shown in the readme?  https://github.com/KaufHA/kauf-rgbww-bulbs#readme

Maybe try doing a factory reset as described here: https://github.com/KaufHA/kauf-rgbww-bulbs#readme

Do you have some automation doing this?  I can't replicate here.  Maybe try removing from Home Assistant and seeing if it still does it.

arfletcher, 2023-06-20:
My yaml file is the same except for the name. See below:

substitutions:
  name: bedroom-lamp-a
  friendly_name: Bedroom Lamp A

packages:
  kauf.rgbww: github://KaufHA/kauf-rgbww-bulbs/kauf-bulb.yaml

wifi:
  ssid:     !secret ssid
  password: !secret wifi

I haven't set up any automations for any of these bulbs yet. 

In fact I just connected the last one to HA without any flashing and it does the same thing. Orange.

I used the /reset URL and the bulb was reset. Using the bulb's web page it will turn on as white. When I turn it on via HA it turns on orange. Then the bulb webpage turns on as orange. 

So HA is overwriting something to make it orange. Im at a loss as to what that is.

arfletcher, 2023-06-20:
A fresh HA instance doesnt show any issues - the last colour setting is remembered perfectly. I'll look to my HA configs for the issue.

Thank you for your help.

arfletcher, 2023-06-20:
FYI: I had a light_profiles.csv file containing an "all lights" default that was overriding bulb colour when HA turned on the lights.

bkaufx, 2023-06-20:
Weird I've never heard of that.  Thanks for letting me know.  Glad you got it sorted out.