KaufHA/kauf-rgbww-bulbs #41: A15 Bulb Flashing after OTA update

funkboxing, 2023-10-05:
I have four kauf bulbs: an A15, two A21s and one BR30. I went to change them to the minimal yaml and the A21 and BR30 took the OTA update just fine and worked again afterwards. The A15 completed the upload, but couldn't reconnect and immediately started flashing white about 3x per second. Now flashing is all it does. I've tried rebooting 5x but no effect, the AP\captiveportal doesn't start and it never connects to wifi. I haven't opened it up and tried FTDI yet, not sure there's any point in that but thought I'd check here.

Is this bulb bricked or is there anything left worth trying?

bkaufx, 2023-10-09:
If you return via amazon, put a note in the box with your username and that you posted on github so I can remember.  Thanks.

r3pwn, 2024-01-19:
I just recently encountered the same issue when trying to go from a "full" build to a "minimal" build on my A15, as well. I'm also outside the return window, but would be happy to disassemble mine and provide any dumps, or try any experiments

rainiur, 2024-04-29:
Just did a minimal flash on mine and having the same problem. Is there any way to fix this?