KaufHA/kauf-rgbww-bulbs #42: 4 Bulbs stopped responding after latest firmware update to 2023.9.3

Makr91, 2023-10-08:
After updating ESPhome, I went through an updated all entities. I did not make any changes to the bulbs configurations.

The configs are as follows:

```
substitutions:
  disp_name: "office-ceiling-fan-light"
  friendly_name: Office Ceiling Fan Light
  update_time: 10s
  max_temp: "100.0"

  sub_red_pin:    GPIO4
  sub_green_pin:  GPIO12
  sub_blue_pin:   GPIO14
  sub_cw_pin:     GPIO5
  sub_ww_pin:     GPIO13

## Connection Configuration
wifi:
  domain: !secret domain
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
    - ssid: !secret wifi_ssid2
      password: !secret wifi_password2
    - ssid: !secret wifi_ssid3
      password: !secret wifi_password3

  ap:
    ssid: "${friendly_name}"
    password: !secret fallback_password

logger:
  level: DEBUG

api:
  encryption:
      key: !secret api_key

ota:
  password: !secret ota_password

time:
  - platform: sntp
    id: sntp_time   
    timezone: !secret timezone
    servers:
      - !secret ntp_server1
      - !secret ntp_server2

captive_portal:

web_server:
  local: true

## Board Configuration
esphome:
  name: ${disp_name}
  name_add_mac_suffix: false

esp8266:
  board: esp01_1m
  restore_from_flash: true

## Sensor and Component Configuration
button:
  - platform: restart
    id: restart_button
    name: $friendly_name Restart Firmware
    entity_category: diagnostic

sensor:
  - platform: uptime
    name: $friendly_name Uptime
    update_interval: 60s
    entity_category: diagnostic
  - platform: wifi_signal
    name: "${friendly_name} WiFi Signal"
    update_interval: 60s
    entity_category: diagnostic

text_sensor:
  - platform: wifi_info
    ip_address:
      name: $friendly_name IP Address
      entity_category: diagnostic

output:
  - platform: esp8266_pwm
    pin: $sub_red_pin
    frequency: 200 Hz
    id: pwm_red
    max_power: .85
  - platform: esp8266_pwm
    pin: $sub_green_pin
    frequency: 200 Hz
    id: pwm_green
    max_power: .85
  - platform: esp8266_pwm
    pin: $sub_blue_pin
    frequency: 200 Hz
    id: pwm_blue
    max_power: .85
  - platform: esp8266_pwm
    pin: $sub_cw_pin
    frequency: 400 Hz
    id: pwm_cw
    max_power: .85
  - platform: esp8266_pwm
    pin: $sub_ww_pin
    frequency: 400 Hz
    id: pwm_ww
    max_power: .85

light:
  - platform: rgbww
    restore_mode: ALWAYS_ON
    name: "${friendly_name}"
    effects:
      - random:
      - strobe:
      - flicker:
      - pulse:
      - pulse:
          name: "Fast Pulse"
          transition_length: 0.5s
          update_interval: 0.5s
      - pulse:
          name: "Slow Pulse"
          transition_length: 1s
          update_interval: 2s
    red: pwm_red
    green: pwm_green
    blue: pwm_blue
    cold_white: pwm_cw
    warm_white: pwm_ww
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
    constant_brightness: true
    color_interlock: true
```

Makr91, 2023-10-08:
I opened the bulbs up, pried out the circuit board, detached the control board from the LED panel, and flashed it using ESPhome (web).

I am assuming that since some of these lights also have a Shelly controlling the whole fixture, that the shelly restarted in the middle of their updates or something?

closed 2023-10-08 by Makr91

bkaufx, 2023-10-09:
That's highly possible.  ESP8266 chips are pretty sensitive to being reset during the update process.  Once the firmware is uploaded the chip still has to apply the update and it can sometimes take up to a minute.

Makr91, 2023-10-09:
Do you have any recommended way to get the LED panel detached from bulb? while I didn't destroy the ones I have fixing them,  two did get a notch in them.  I assume a sharp exacto blade to cut the sealant, and then a hardened steel pick that you would dig under the edges about 90*s counter clockwise if the headers are south?