KaufHA/kauf-rgbww-bulbs #43: captive portal compile error ESPHome 2023.10.0

SystemIdleProcess, 2023-10-18:
Update updated ESPHome in HA and went to update Kauf bulbs (A21 & BR30) and RGB Wall Switch and received the following error.
I'm no developer but looking at ESPHome captive_portal.cpp (https://github.com/esphome/esphome/blob/19736f6e537269fff2ec42b0348b43b277dc8fb2/esphome/components/captive_portal/captive_portal.cpp#L47) and KaufHA common (https://github.com/KaufHA/common/blob/3f64a5da38cd1853dd66f9963d4e822e61e714de/components/captive_portal/captive_portal.cpp#L72) it looks like ESPHome has changed a data type and it's having issues converting/casting it.

NFO ESPHome 2023.10.0
INFO Reading configuration /config/esphome/kauf-bulb-141.yaml...
INFO Updating https://github.com/KaufHA/kauf-rgbww-bulbs@None
INFO Generating C++ source...
INFO Compiling app...
Processing kauf-bulb-141 (board: esp01_1m; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/kauf-bulb-141/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling .pioenvs/kauf-bulb-141/src/esphome/components/light/light_call.cpp.o
Compiling .pioenvs/kauf-bulb-141/src/esphome/components/light/light_json_schema.cpp.o
Compiling .pioenvs/kauf-bulb-141/src/esphome/components/light/light_output.cpp.o
src/esphome/components/captive_portal/captive_portal.cpp: In member function 'void esphome::captive_portal::CaptivePortal::start()':
src/esphome/components/captive_portal/captive_portal.cpp:76:37: error: invalid cast from type 'esphome::network::IPAddress' to type 'uint32_t' {aka 'unsigned int'}
   76 |   this->dns_server_->start(53, "*", (uint32_t) ip);
      |                                     ^~~~~~~~~~~~~
Compiling .pioenvs/kauf-bulb-141/src/esphome/components/light/light_state.cpp.o
Compiling .pioenvs/kauf-bulb-141/src/esphome/components/logger/logger.cpp.o
*** [.pioenvs/kauf-bulb-141/src/esphome/components/captive_portal/captive_portal.cpp.o] Error 1
src/esphome/components/light/light_state.cpp: In member function 'void esphome::light::LightState::wled_apply()':
src/esphome/components/light/light_state.cpp:220:14: error: no match for 'operator[]' (operand types are 'esphome::network::IPAddress' and 'int')
  220 |     if ( addr[3] >= 254 ) {
      |              ^
src/esphome/components/light/light_state.cpp:226:9: error: no match for 'operator[]' (operand types are 'esphome::network::IPAddress' and 'int')
  226 |     addr[3]++;
      |         ^
src/esphome/components/light/light_state.cpp:271:14: error: no match for 'operator[]' (operand types are 'esphome::network::IPAddress' and 'int')
  271 |     if ( addr[3] + packet1_length >= 255 ) {
      |              ^
src/esphome/components/light/light_state.cpp:274:11: error: no match for 'operator[]' (operand types are 'esphome::network::IPAddress' and 'int')
  274 |       addr[3] += packet1_length;
      |           ^
*** [.pioenvs/kauf-bulb-141/src/esphome/components/light/light_state.cpp.o] Error 1
========================== [FAILED] Took 5.97 seconds ==========================

bkaufx, 2023-10-18:
Should be fixed now.  Had to update the custom components for the latest ESPHome release.  In order for you to see the fix you'll need to delete your .esphome/packages folder or wait 24 hours for ESPHome to check for an updated package again.

teamRibcage, 2023-10-18:
That corrected the errors for me! (4x BLF10's & 1x BLF15)

Thanks for the quick fix, and for a great line of products in general. These bulbs have been fantastic in our home - looking forward to adding a few more! Keep up the great work!! 🙂

SystemIdleProcess, 2023-10-19:
I'm using the HA add-on for esphome so couldn't find a way to clear out packages folder. But this morning it showed the update button and it was able to compile and push without error. Thank you

closed 2023-10-19 by SystemIdleProcess