KaufHA/kauf-rgbww-bulbs #44: ESPHOME 2023.10.5 update fail to Bulb

bert269, 2023-11-01:
A few releases ago, it was the same problem. Then it work for about two releases and now updates fails again:

[image]

This is runing from within HAOS dashboard/UI.

bert269, 2023-11-02:
"run the update from teh dashboard" - which dashboard are you referring to?
ESPHOME dasdhboard looks like this:
[image]
There is no option to do update.

Click on 'visit' shows that only 10.4 is available:
[image]

The latest on Github is still only 1.92 (2023.9.1) !

When I try to do the upgrade from HAOS notification area, I get the same message as in opening comment.

bert269, 2023-11-02:
ESPHOME 10.6 was just realeased this morning - same issue.

bert269, 2023-11-03:
This is the log from HAOS:
'''Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:230
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: November 2, 2023 at 5:52:21 AM (9 occurrences)
Last logged: 7:06:42 AM

[546621103552] Error updating kitchen-lamp.yaml via OTA; Try again in ESPHome dashboard for more information.
[546375046592] Error updating kitchen-lamp.yaml via OTA; Try again in ESPHome dashboard for more information.
[546525463360] Error updating kitchen-lamp.yaml via OTA; Try again in ESPHome dashboard for more information.
[546525463360] Error updating foyer-pres.yaml via OTA; Try again in ESPHome dashboard for more information.
[546523563328] Error updating kitchen-lamp.yaml via OTA; Try again in ESPHome dashboard for more information.
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 230, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 2035, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2072, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 876, in entity_service_call
    response_data = await _handle_entity_call(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 948, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 157, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 432, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/esphome/update.py", line 179, in async_install
    raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Error updating kitchen-lamp.yaml via OTA; Try again in ESPHome dashboard for more information.

Also, during the update process, when I 'visit' the bulb from ESPHOME, I do see the following on the web page:
[image]

But the firmware does not get updated to the device. It is still stuck on 10.4.

bkaufx, 2023-11-03:
Were these previously updated fine through HA or the dashboard?  If the devices aren't showing up as online in the dashboard then that is probably the issues.  Since the ESPHome dashboard thinks the devices are up to date it is not showing the update option in that view.  If you press the three dots you can click "install" and its the same thing.

You might need to point the yaml configs to the devices' IP addresses using use_address or set a manual IP.

bert269, 2023-11-04:
Yes. All updated through HA - that's how I got it to 10.4.
None of my working ESP devices ever shows up as online in ESPHOME, but the
work just fine. Thats not the problem.
Pressing the three dots, install, wirelessly has the same effect - does not
update this bulb.

My devices have static ips in my router table. I will try to add it into
the yaml code for this bulb

Thanks.

bert269, 2023-11-04:
This bulb already had manual IP address assigned in yaml code.
Now I get this memory error trying to compile and install the code.

bkaufx, 2023-11-06:
When you try to update through the dashboard, what error do you get there?

bert269, 2023-11-06:
Sorry. I threw the bulb out. Tired of battling with it.

closed 2026-10-29 by KaufHA