KaufHA/kauf-rgbww-bulbs #46: Can't connect lights after HA ESPHome update

trovador, 2023-11-13:
Hi, I updated ESPHome on HA to 2023.10.6. Using 8 Kauf A21 bulbs. They were working fin until then, and since they have not responded. Tried resetting and sometimes I've been able to see the Bulb KB X 1.921 wifi show up but I can't seem to connect to it in this main room where everything is setup. I pulled that bulb out to another place and I can connect to it there. So my guess is wifi issues, but the weird thing is I haven't made any changes to the wifi at all, and it all just stopped working after the HA ESPHome update. 

I tried updating the bulb outside of the space and  keeping the hard coded wifi credentials to my setup place. Moving over they just won't connect. ,Not sure if there is something different in the new firmware that may be affecting the wifi in some way?

Open to ideas on troubleshooting this setup which is in production at a local yoga studio where I live. Help!

trovador, 2023-11-16:
So I was able to identify that it just doesn't work with the Hard-Coded SSID. Not sure why, perhaps issue with spaces in the SSID name, or not picking up the password. THis happened with the firmware I flashed via ESPHome on my local laptop which has the info in secrets.yaml file. TEsting out with flashing to the stock firmware version.

xanderphillips, 2023-11-29:
I have a similar issue.  I had to serial flash to older version to get bulb to stay on and not keep rebooting it's AP, Connected it to ESPHome and it showed up in Home Assistant just fine.  I tried to push update to YAML to the bulb to enable web_server and that automagically updated the firmware back to latest, which again breaks it from connecting to my wifi and watching wifi broadcasts it comes up as an AP for a while, then goes away over and over.  1.92 is broken for me, but 1.87 worked, but I didn't mean to update to 1.92.  I am new enough to ESPHome to not know how to prevent it from doing the update.

trovador, 2023-11-29:
Turns out I had a weird issue where it wouldn't pick up the hard-coded Wifi credentials for some reason. I actually took the bulbs out and went to another network SSID and they worked. So I then updated to latest firmware and added multiple networks with all the SSID's I use and now it's working. Still I won't be updating ESPHome in HA again.. it was a pain.

xanderphillips, 2023-11-29:
> Turns out I had a weird issue where it wouldn't pick up the hard-coded Wifi credentials for some reason. I actually took the bulbs out and went to another network SSID and they worked. So I then [...]

Not sure why 1.92 breaks mine, but it does.  Searching now for a way to disable ESPHome OTA for this bulb so I don't accidentally break it again as even with a jig re-flashing it over serial is a pain.