KaufHA/kauf-rgbww-bulbs #47: Turn bulb back on after DDP packets stop being sent

rsather70, 2023-11-15:
I am using these bulbs in my Christmas light show, and when the show is over the bulbs sit in an 'off' state (powered on but no light).  Is there a command I can send from Falcon Player at end of show to trigger the bulb to go back to standard 'on' mode or does it require the light switch to be toggled?  If this is not a current option can it be added as a future enhancement request?

bkaufx, 2023-11-15:
I have a beta of this working.  If you add the following to your local config in ESPHome dashboard it should time out after 10s of not receiving a DDP packet and return to the previously set value.  Let me know if it works I haven't been able to test it yet.

   - source:
      type: git
      url: https://github.com/bkaufx/kauf-rgbww-bulbs
    refresh: always

rsather70, 2023-11-17:
I don’t use esphome.  My smart home setup is Hubitat so these bulbs aren’t integrated.  I do have home assistant spun up that I’m just playing with I could try it there especially if it helps you test since I’m about to do this daily with my Christmas show - no setting on the bulb itself though?-Ryan SatherPlease excuse my typos from my mobile device Sent from my iPhone

bulldozer2003, 2023-11-18:
On a similar note, when the bulb is in an off state and a DDP packet is received the bulb will light up but remains in an off state both in the UI and HA.

I've got a WLED controller on a timer that runs from 5-10pm. I don't mind the bulbs staying on all night in their last DDP commanded color but wanted to have HA turn the bulbs off at sunrise. HA thinks they're off because the bulbs reports it is off.

brennja, 2024-10-10:
Regarding DDP timeout, you sent my colleague 'kauf-bulb-ddp-timeout-beta.bin.gz" that works.
I don't have ESPHome setup.
Could DDP timeout be added to latest UI?