KaufHA/kauf-rgbww-bulbs #48: Can't update bulbs with ESPHome 2023.11.4

mccmax1395, 2023-11-24:
When I try to update my bulbs with ESPHome 2023.11.4, I get the following errror `ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)`

My configuration is pretty minimal expect for the kauf package.  This is what it looks like:
```
substitutions:
  name: "bedroom-bulb-1"
packages:
  Kauf.RGBWW: github://KaufHA/kauf-rgbww-bulbs/kauf-bulb.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
api:
  encryption:
    key: X1py3+1iy1s4Jkwzh6B6bu6iIlfEtbOIEdNTw2J39Q0=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

mhump711, 2023-11-26:
I had the same problem. 

If you change the package to the following it should work:
`packages:
  Kauf.RGBWW: github://KaufHA/kauf-rgbww-bulbs/kauf-bulb-lite.yaml`

Note you will lose some features but the ones I care about still work

mccmax1395, 2023-11-26:
Thanks for the suggestion, that worked for me.  I also updated back to 
`packages: Kauf.RGBWW: github://KaufHA/kauf-rgbww-bulbs/kauf-bulb.yaml` afterwards which sort of worked.  Sometimes it would work and sometimes I would get the same error.  I just went through the process a couple times and I eventually got all my bulbs updated.  

I'll close this issue since I was able to update all my bulbs but I have a feeling this might become an issue again down the road.

closed 2023-11-26 by mccmax1395