KaufHA/kauf-rgbww-bulbs #49: Unable to update due to lack of space

mtgpcs, 2023-12-05:
INFO Uploading /data/build/kauf-bulb-xxxxx/.pioenvs/kauf-bulb-xxxxxx/firmware.bin (606672 bytes)
INFO Compressed to 417949 bytes
ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)

The only workaround I found was to use the kauf-bulb-minimal.yaml then to change it back after and update it again. This was fine then the next ESPHome update same issue. 

This is happening across all my Kauf bulbs.

bkaufx, 2023-12-05:
What do you have in the yaml besides the defaults?  This never used to be a problem for the bulbs so I'll have to take a look and see what changed from the ESPHome side.  You can always just take out the API encryption.

mtgpcs, 2023-12-06:
substitutions:
  name: kauf-bulb-xxxxx
packages:
  Kauf.RGBWW: github://KaufHA/kauf-rgbww-bulbs/kauf-bulb.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
api:
  encryption:
    key: 

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

This is my whole yaml. What does taking out the encryption key do?

SpencerDub, 2023-12-06:
I'm also having this issue. My YAML is also fairly minimal (a few things modified for privacy):

```
substitutions:
  name: spouses-bedside-light-2
  friendly_name: "Spouse's Bedside Light 2"

packages:
  Kauf.RGBWW: github://KaufHA/kauf-rgbww-bulbs/kauf-bulb.yaml

esphome:
  name: ${name}
  name_add_mac_suffix: false
  comment: The bulb in [Spouse]'s nightstand lamp that points away from the bed.

api:
  encryption:
    key: [encryption key]

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: XXX.XXX.XXX.XXX
    # Set this to the IP address of the router. Often ends with .1
    gateway: XXX.XXX.XXX.XXX
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0
```

bkaufx, 2023-12-06:
You may need to remove and re-add into home assistant after reflashing without api encryption.

olivervhansen, 2023-12-07:
Removing the api: / encryption: / key: lines and then installing the config worked for me.
I did need to remove the device from Home Assistant and then re-add it. Wasn't sure how to do that at first but I did it by: **Settings -> Devices -> ESPHome -> 3 dot menu next to the device and choose delete**.

Waited a few minutes and in one case rebooted the light and then it was detected by Home Assistant.

epyonavenger, 2024-04-02:
Had the same issue, and removing the api encryption key also solved it, but that feels like a hack, not a solution. Ideally, we'd want that to work. Is there something else we can do to make it work? Higher compression? Different library? Other?

niemyjski, 2024-05-02:
Same, I made a pr for this repo and now I can't flash it.

niemyjski, 2024-05-02:
@KaufHA How have you worked around this other than double flashing?

bkaufx, 2024-05-02:
No I will take a look today and see how close the space is to the limit.  There is probably nothing I can do except remove functionality from the main yaml file.  If the size is very close to the threshold I can potentially edit some of the components to remove unused functionality that is not being optimized out by the compiler.

bkaufx, 2024-05-02:
What I'm probably going to have to do is create a separate yaml package specifically for dashboard import that removes a bunch of options that are easily configured in yaml to save space.

If I remove the entities for boot state, max power, default fade, and no hass, then that gets it small enough to fit with API encryption.  Those can all be easily configured with a single substitution each.

bkaufx, 2024-05-02:
Okay, the below packages section imports the yaml I am proposing to use.  I think I will change out the main kauf-bulb.yaml config file so that this change will go out to everyone that has imported into their dashboard.  It will be a breaking change for anyone depending on those config entities.  However, I like that better than creating a whole other yaml file for no real reason.  Anyone installing a binary file update will still get all the old config entities and no API encryption.

Test out the below code if you feel like it.  If I don't hear anything in the next day or two I'm just going to do it.

```
packages:
  kauf.rgbww: github://bkaufx/kauf-rgbww-bulbs/kauf-bulb.yaml
```

bkaufx, 2024-05-02:
With that yaml, you will still have to flash twice to get on that track.  The current yaml does not leave enough space for the new yaml.  However, once you have the new yaml programmed then there will still be enough space to re-flash the new yaml.

closed 2026-10-29 by KaufHA