KaufHA/kauf-rgbww-bulbs #50: ignoring ddp packet with non zero data offset

ridl8349, 2023-12-11:
Using 5 of the A21 sized bulbs in a Christmas xLights setup. Worked last year on v1.85. Updated all 5 bulbs to v1.92 and all 5 are showing the error referenced in the v1.86 in the verbose logs. xLights is setup according to YouTube guide. 

xLights version 2023.20

I saw and read the reference in the release notes, but do not understand how it is affecting the DPP protocol over the network coming from xLights.

I did verify no other DDP is being broadcasted over the network All other functions of the bulb for correctly.

bkaufx, 2023-12-12:
What is the error message?

bkaufx, 2023-12-12:
oh data offset?  I think there is a setting in xlights to make it zero.

ridl8349, 2023-12-12:
> What is the error message?

The only thing in the log is "ignoring DDP with non-zero data offset". There there is a bunch of random numbers and letters.

bkaufx, 2023-12-12:
Try "Keep Channel Numbers" in the xlights config.

ridl8349, 2023-12-12:
[image]

Keep channel numbers is selected for all bulbs.

ridl8349, 2023-12-12:
[image]
Here is the log from the bulb. 

It starts as soon as I press output to lights.

ridl8349, 2023-12-14:
Update, if I roll the bulb back to v1.85 it functions with xlights with problems. v1.86 and up seem to have this issue. 

[image]

cpuchip, 2024-11-22:
I was running into the same issue running
Firmware version 1.93(u) made using [ESPHome](https://esphome.io/) version 2024.4.2.
and xLights 2024.17, bulbs are pared into Home Assistant and I have them working there, I've enabled DDP on the bulbs, and get a similar error message, if I uncheck "Keep Channel Numbers" then they work, but xLights throws a connect error. I haven't figured that out yet, but I can output to the lights now in test and in my models

kjans, 2024-11-22:
I had the same issue running 1.93...
Unchecking "Keep Channel Numbers" appears to fix it... 
I will do more testing to be sure ...And I don't get a "connect error"...

cpuchip, 2024-11-22:
I cleared up my connection error, I had a controller with no IP for ddp listed in my controllers tab in xLights. So look good to me, I wont know about any latencies until I get my other pixel lights up.