KaufHA/kauf-rgbww-bulbs #56: How can I remove encryption on an already configured bulb?

HarleyGaniere, 2024-02-13:
Howdy, I bought a four pack of bulbs off of Amazon and three of the bulbs worked perfectly fine. However one I cannot figure out how to remove encryption on. I believe it was previously setup by someone then returned and Amazon sent me a 'new' open boxed bulb. Which I wouldn't mind if I could figure out how to get it reset as it seems to power on just fine.

My method of setup has been
Step 1: plug bulb in
Step 2: turn on/off light switch a few times until it would light up yellow for a few seconds.
Step 3: connect to Wi-Fi network of the bulb
Step 4: get redirected and put in Wi-Fi credentials.
Step 5: go into home assistant and wait until the bulb is discovered.

All works well afterwards for three of the bulbs aside from the fourth.

For the fourth bulb, I follow the same steps above but am greeted by a message in home assistant requesting the encryption key of the bulb. Enclosed below is a photo of the encryption key request.
[image]

Any suggestions to reset the bulb?

I've attempted to 1. turn the light bulb on and off a bunch of times until it turns red then re-enter pairing mode. and 2. reflash OTA updates with the 1.92 .bin update file. Which is the only thing I can think of to fix this, but it doesn't work so I have to for sure be doing something wrong.

Any help would be greatly appreciated.

Kind regards,
Harley Ganiere

bkaufx, 2024-02-13:
Reflashing doesn't work like it won't reflash properly or it reflashes but still wants an encryption key?  Reflashing should remove encryption.  If you are having trouble with the reflashing process I'd probably try curl to upload the bin.gz file or I'd do it through the ESPHome dashboard.

`curl -v -X POST -H "Content-Type: multipart/form-data" -F update=@firmware_filename.bin.gz  "http://192.168.1.16/update"`

HarleyGaniere, 2024-02-14:
Thank you for the instructions @bkaufx. Your instructions were exactly what I needed. I was having trouble reflashing over the ESPHome Dashboard so the curl command worked perfectly.

If anyone experiences this issue and stumbles across this GitHub posting, follow the steps below to fix.

Step 1: Make sure your bulb is connected to your Wi-Fi network and write down its IPv4 address.
Step 2: Open Home Assistant with ESPHome installed
Step 3: Open ESPHome
Step 4: Create new device
Step 5: Name device appropriate name "kauf-bulb or kauf-bulb-room, etc."
Step 6: Select ESP8266
Step 7: Skip installation
Step 8: Click edit on the new device
Step 9: Copy and paste the kauf-bulb.yaml from this repository into your devices .yaml file, change the lines that include the name to reflect your desired device's name if needed, then save.
Step 10: Click the triple dot menu
Step 11: Click install
Step 12: Select Manual Download (A .bin file should download, it's way too big to be used so keep on following the steps)
Step 13: Download and open PeaZip (7-Zip/WinRAR probably work fine for this too)
Step 14: Convert the .bin file into a GZip (.gz) file. The file should read something along the lines of "kauf-bulb.bin.gz"
Step 15: Execute bkaufx's curl command of `curl -v -X POST -H "Content-Type: multipart/form-data" -F update=@firmware_filename.bin.gz  "http://192.168.1.16/update"` in the folder where the firmware is located. Matching the @firmware_filename with the name of your .bin.gz file ("kauf-bulb.bin.gz" by default) and the "16" to your bulbs IPv4 address.
Step 16: Reconfigure the bulb with your network credentials. 
Step 17: ???
Step 18: Profit

Note: Flashing with the OTA option in the initialized webpage on "192.168.4.1" did not work for me, neither did flashing over ESPHome. The curl command was the only method I've tried that worked.

closed 2024-02-14 by HarleyGaniere