KaufHA/kauf-rgbww-bulbs #64: Component not found: button ESPHOME 2024.6.1

m1k3n34l, 2024-06-23:
INFO ESPHome 2024.6.1
INFO Reading configuration /config/esphome/kauf-bulb-7c3272.yaml...
INFO Updating https://github.com/KaufHA/kauf-rgbww-bulbs@None
ERROR Unable to load component button:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 176, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/esphome/esphome/components/button/__init__.py", line 48, in <module>
    cv.ENTITY_BASE_SCHEMA.extend(web_server.WEBSERVER_SORTING_SCHEMA)
                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'esphome.components.web_server' has no attribute 'WEBSERVER_SORTING_SCHEMA'
Failed config

button: [source /data/packages/0cf47246/kauf-bulb.yaml:114]
  
  Component not found: button.
  - platform: restart
    id: restart_button
    name: Bar Back Can Left Restart Firmware
    entity_category: diagnostic
    disabled_by_default: true

bkaufx, 2024-06-24:
You might not have our latest yaml.  Maybe clean build files or delete the .esphome/packages folder manually.

teamRibcage, 2024-06-24:
I was actually getting the same error myself until a few updates got pushed to the ESPHome repositories last evening. Make sure your Container/ESPHome files and `.yaml` configurations are up to date and give it another go. Everything was compiling correctly for me again as of around 20:00 EDT yesterday.

May or may not be your issue, but just in case here's my `.yaml` for my BLF10 Bulbs. Good luck! 🙂
```
esphome:
  comment: >
    Den Lamp #1 - KAUF Smart Bulb Model: BLF10

substitutions:
  name: den-lamp-1
  friendly_name: Den Lamp 1
  light_restore_mode: RESTORE_AND_OFF  # Set Default State at Power-On - Always Off

packages:
  kauf.rgbww: github://KaufHA/kauf-rgbww-bulbs/kauf-bulb.yaml

api:
  encryption:
    key: !secret den_lamp1_key

ota:
  - platform: esphome
    password: !secret den_lamp1_ota

safe_mode:
  disabled: false

wifi:
  ssid: !secret den_ssid
  password: !secret den_password
  # use_address: 192.168.xx.xx

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "DenLamp1AP"
    password: !secret recovery_password
```

m1k3n34l, 2024-06-25:
Interesting; Clean Build failed in the same manner, could find no local cache of Kauf yaml files, same results after HA restart, then suddenly started working overnight.  Didn't realize the ESPHOME repos were updated between builds, but it seems to have self-resolved (now on to ESPHOME 6.2...)

closed 2024-06-25 by m1k3n34l