KaufHA/kauf-rgbww-bulbs #69: Bulb reboots randomly and sometimes changes to the default green

lexi-the-cute, 2024-09-11:
I've been trying to figure out why it keeps resetting and attempting to resolve it. It does seem to happen less often when not using the DDP / WLED effect

this is my ESPHome config

```yaml
# Device Unique Settings (requires either the kauf config linked below or passing the vars ourselves)
substitutions:
  name: lexi-room-bulb-1
  friendly_name: "Lexi's Room Bulb 1"
  ssid: "Lexi's Room Bulb 1 Hotspot"
  sub_max_power: '1.0'
  light_restore_mode: ALWAYS_ON
  mqtt_prefix: iot/$name

# Grab Recommended Kauf Config
# https://github.com/KaufHA/kauf-rgbww-bulbs/blob/main/kauf-bulb.yaml
packages:
  Kauf.RGBWW: github://KaufHA/kauf-rgbww-bulbs/kauf-bulb.yaml

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_encryption_key

# MQTT Server
mqtt:
  broker: !secret mqtt_broker
  topic_prefix: $mqtt_prefix
  reboot_timeout: 0s
  username: lexi
  password: !secret lexi_mqtt_password

# Set OTA Password
ota:
  - platform: esphome
    password: !secret ota_password

# Connect to Wifi or Fallback to Hotspot
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: $ssid
    password: !secret hotspot_fallback_password

# Component to report useful stats on startup
debug:

# Sensor configured to read text information from the debug component
text_sensor:
  - platform: debug
    device:
      name: "Device Info"
    reset_reason:
      name: "Reset Reason"

# Sensor configured to read number information from the debug component
sensor:
  - platform: debug
    free:
      name: "Heap Free"
    fragmentation:
      name: "Heap Fragmentation"
    block:
      name: "Heap Max Block"
    loop_time:
      name: "Loop Time"

# Sets logger to debug verbosity (the default)
logger:
  level: debug
```

MQTT is locked down so you have to be authenticated to read or write to the topics, `homeassistant/#`, `esphome/#`, and `iot/#`. the iot topic is where i have my bulb prefixed to with the command topic being `iot/lexi-room-bulb-1/light/lexi_s_room_bulb_1/command`

This is the latest device info from the debug component above

```
2024.8.3|Flash: 1024kB Speed:40MHz Mode:DOUT|Chip: 0x00303190|SDK: 2.2.2-dev(38a443e)|Core: 3.1.2|Boot: 31|Mode: 1|CPU: 80|Flash: 0x00146085|Reset: External System|External System	
```

[image]

When I still had DDP enabled, it was causing segfaults. I'm currently trying to get QEMU running with ESP8266 (and also will be messing with ESP32, but that's more for wanting to create my own devices in the semi-distant future to do custom things) so I can get a more in depth look as to the cause of the random reboots

The reboots happen anywhere from a few minutes to few hours. This issue has stuck around through multiple builds and changes of the ESPHome config, so it isn't likely to just be a bad build, but something else. Potentially corrupted data outside of the firmware, but I currently have no way to check that

lexi-the-cute, 2024-09-11:
This is the exact bulb I bought https://a.co/d/6JRLZgg

lexi-the-cute, 2024-09-11:
It just rebooted again while I was closing the issue (with the default green)

Device Info:
```
2024.8.3|Flash: 1024kB Speed:40MHz Mode:DOUT|Chip: 0x00303190|SDK: 2.2.2-dev(38a443e)|Core: 3.1.2|Boot: 31|Mode: 1|CPU: 80|Flash: 0x00146085|Reset: External System|External System
```

[image]

lexi-the-cute, 2024-09-11:
What's frustrating is, I bought 2 bulbs at once. One bulb just stopped working entirely out of the blue and this one has this problem. Hopefully the replacement for the completely broken bulb works fine and that it's just a bad batch

lexi-the-cute, 2024-09-13:
it may have been `light_restore_mode: ALWAYS_ON`. will have to do further tests

lexi-the-cute, 2024-09-13:
i think the first two bulbs were just a bad batch. bulb 1 is still misbehaving, but bulb 2's replacement seems fine

lexi-the-cute, 2024-09-13:
looks like disabling `light_restore_mode: ALWAYS_ON` just masks the issue. the home assistant logs shows that the bulb still reboots itself due to "External System" even when it visually doesn't look like it changes. the replacement bulb for the one that just straight up died doesn't appear to be having any issues according to the home assistant logs. looks like i just got 2 bad bulbs in the same order. leading me to think the bulbs were not designed well when it was just really bad luck

lexi-the-cute, 2024-09-14:
*sigh*

[image]
[image]
[image]
[image]

lexi-the-cute, 2024-09-14:
in my attempt to debug this in QEMU, i attempted to convert this over from ESP8266 Wi-Fi to ESP32 Ethernet and came up with this cursed config. it fails to compile due to expecting the header file for ESP8266 Wi-Fi

```yaml
# Device Unique Settings (requires either the kauf config linked below or passing the vars ourselves)
substitutions:
  name: lexi-kauf-qemu-1
  friendly_name: "Lexi's Kauf QEMU 1"
  ssid: "Lexi's Kauf QEMU 1 Hotspot"
  sub_max_power: '1.0'
  #light_restore_mode: ALWAYS_ON
  mqtt_prefix: iot/$name
  forwarded_address: 192.168.2.122  # Used for port forwarding

# Grab Recommended Kauf Config
# https://github.com/KaufHA/kauf-rgbww-bulbs/blob/main/kauf-bulb.yaml
packages:
  Kauf.RGBWW: github://KaufHA/kauf-rgbww-bulbs/kauf-bulb.yaml

# https://esphome.io/components/esp8266.html
esp8266: !remove

# https://github.com/platformio/platformio-docs/blob/develop/boards/espressif32/esp32doit-devkit-v1.rst
# https://esphome.io/components/esp32.html
esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf

# Installs PR with open_eth driver
external_components:
  - source: github://pr#7020
    components: [ ethernet ]

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_encryption_key

# MQTT Server
mqtt:
  broker: !secret mqtt_broker
  topic_prefix: $mqtt_prefix
  reboot_timeout: 0s
  username: lexi
  password: !secret lexi_mqtt_password

# Set OTA Password
ota:
  - platform: esphome
    password: !secret ota_password

# Replace Output
# https://esphome.io/components/output/
# https://esphome.io/components/output/esp8266_pwm.html
# https://esphome.io/components/output/ledc.html#esp32-ledc-output
output:
  - id: !remove pwm_red
  - id: !remove pwm_green
  - id: !remove pwm_blue
  - id: !remove pwm_cw
  - id: !remove pwm_ww
  - platform: ledc
    pin: $sub_red_pin
    frequency: $sub_red_freq
    id: pwm_red
    max_power: $sub_max_power
  - platform: ledc
    pin: $sub_green_pin
    frequency: $sub_green_freq
    id: pwm_green
[... 133 lines cut ...]
text_sensor: !remove
  - platform: debug
    device:
      name: "Device Info"
    reset_reason:
      name: "Reset Reason"

# Sensor configured to read number information from the debug component
sensor:
  - platform: debug
    free:
      name: "Heap Free"
    # fragmentation:
    #   name: "Heap Fragmentation"
    block:
      name: "Heap Max Block"
    loop_time:
      name: "Loop Time"
    # psram:
    #   name: "Free PSRAM"

# Sets logger to debug verbosity (the default)
logger:
  level: debug
```

lexi-the-cute, 2024-09-20:
so, at this point, i just hate these lights. even the replacement for bulb 1 is resetting on me and im just like, why? like it shouldn't be happening this much. these are 3 bulbs out of 4 that have turned out to be bad. only 1 of them has worked properly and they have been running the same ESPHome firmware

kythre, 2025-01-28:
So far I've had 5 bulbs start doing this and I bought 9. I think I'm gonna have to switch to something else