KaufHA/kauf-rgbww-bulbs #73: Custom Power On State?

aeleus, 2024-11-19:
I just added 4 BRF30 bulbs to Home Assistant. The Power On State options seem to be limited to Off, On - Bright White, or some version of the last value. Is there a way to configure them so that they always power on at, for example, 50% brightness and 2800K? 

Having them come at "last value" doesn't work because I have no control over what state they may be in when someone turns off the switch. Power on at Bright White is no good either.

The bulbs came with firmware version 1.873(f), and so far I have updated 2 of them to 1.93(u).

HA 2024.11.2 running in a container.
I am not running ESPHome Dashboard.
I have not configured the bulbs with any yaml files.

Thanks in advance.

bkaufx, 2024-11-20:
Actually this feature was just added to ESPHome in the new update coming out tomorrow.  For now you'd have to add it to your own yaml.

Later on this week I'll try to get it working and post the yaml.

aeleus, 2024-11-21:
FYI: I flashed one of the Kauf bulbs with the latest Tasmota 14.3.0. It seems Tasmota does not have the ability to configure a user-defined PowerOnState either. :-(

Here are the options for Tasmota:
PowerOnState	Control relay state after powering up the device.
0 / OFF = keep relay(s) OFF after power up
1 / ON = turn relay(s) ON after power up
2 / TOGGLE = toggle relay(s) from last saved state
3 = switch relay(s) to their last saved state (default)
4 = turn relay(s) ON and disable further relay control
5 = after a PulseTime period turn relay(s) ON (acts as inverted PulseTime mode)

bkaufx, 2024-11-23:
For the yaml, here is what you have to add to the default.  Substitution section likely already exists so just add the restore mode substitution to that section.  Change values under initial_state to your liking.

```
substitutions:
  light_restore_mode: ALWAYS_ON

light:
  - id: !extend kauf_light
    initial_state:
      state: on
      brightness: 75%
      color_mode: color_temperature
      color_temperature: 2800K

      # color_mode: rgb
      # red: 100%
      # green: 0%
      # blue: 0%
```

aeleus, 2024-11-23:
Brian,

I cannot figure out where to put the yaml code you gave me. Keep in mind that I have added a bulb to HA by following steps 1 - 4 in the included user manual. I now have the ESPHome integration with a single device (kauf-bulb-b6357f) with an entity named "Kauf Bulb" and an ID of light.kauf_bulb.

The README.MD refers to kauf-bulb.yaml in the context of ESPHome dashboard. I am not running the ESPHome dashboard - do I need to?

Thanks

aeleus, 2024-11-25:
I saw that you updated the README.md and included the yaml code you posted here. I went ahead and installed ESPHome dashboard. With the updated README, ESPHome running, and a rewatch of your YouTube video (ESPHome Dashboard Import for Easy Updates and Renaming), it all makes sense now.

The net result is that I have been able to set a custom power on state!

Thanks for the help.