KaufHA/kauf-rgbww-bulbs #78: Binary size error

eleventhal, 2025-02-07:
I have a new bulb that I was able to import into ESPHome and reflash with updated settings - and I _think_ I used the 1M flash file, but it would seem that perhaps I did not, since I am now getting the following error:

`ERROR Error binary size: Error: ESP has been flashed with wrong flash size. Please choose the correct 'board' option (esp01_1m always works) and then flash over USB.`

Happy to try and reflash, however didn't have an easy way to access the headers, the bulb seems to be sealed :(

I'm assuming this is all user error on my part <sigh>

bkaufx, 2025-02-07:
What bulb size was it?

eleventhal, 2025-02-08:
> What bulb size was it?

A21

bkaufx, 2025-02-08:
Did you just import it automatically or copy-paste in the code from somewhere?

What is the actual yaml you are using?

Does the bulb say (4m) on the side after BLF10?

bkaufx, 2025-02-08:
How long have you had it?

eleventhal, 2025-02-08:
I copy-pasted the code from github. This is my third bulb and I haven't had trouble with the others :/
I can connect to the bulb now and it works, I just can't update the firmware.
I have had it for about 48 hours. 

It goes not say 4m on the side. This is the code I think I uploaded, but in attempting to fix I can't be certain if I imported the 4m package or the 1m package.
 
```
substitutions:
  name: <redacted>
  friendly_name: <redacted>

esphome:
  name: ${name}
  name_add_mac_suffix: false

packages:

  # Bulbs with 1m flash use this package:
  kauf.rgbww: github://KaufHA/kauf-rgbww-bulbs/kauf-bulb.yaml

  # Bulbs with 4m flash use this package instead:
  # kauf.rgbww: github://KaufHA/kauf-rgbww-bulbs/config/kauf-bulb-4m.yaml

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: 192.168.1.134 # only needed for setup
```