KaufHA/kauf-rgbww-bulbs #82: adding extented effects not showing up in webgui

davlaw, 2025-03-20:
effect extension not show up in webgui only WLED / DDP

I added your example;

light:
  - id: !extend kauf_light
    effects:
      - flicker:
          alpha: 94%
          intensity: 12%
      - strobe:
      - pulse:
      - random:

Verification of yaml code, shows its added;

 effects:
  - flicker:
      alpha: 0.94
      intensity: 0.12
      name: Flicker
  - strobe:
      colors:
      - state: true
        duration: 500ms
        brightness: 1.0
        color_brightness: 1.0
        red: 1.0
        green: 1.0
        blue: 1.0
        white: 1.0
        cold_white: 1.0
        warm_white: 1.0
        transition_length: 0s
      - state: false
        duration: 500ms
        brightness: 1.0
        color_brightness: 1.0
        red: 1.0
        green: 1.0
        blue: 1.0
        white: 1.0
        cold_white: 1.0
        warm_white: 1.0
        transition_length: 0s
      name: Strobe
  - pulse:
      transition_length: 1s
      update_interval: 1s
      min_brightness: 0.0
      max_brightness: 1.0
      name: Pulse

bkaufx, 2025-03-20:
Those effects show up somewhere different.  You have to go into the light entity first, and then the effects are part of the light entity.

davlaw, 2025-03-21:
Guess I need to make each bulb id bit more unique.  Not sure I'm seeing enough "effects" for the bulbs in use.   Just started messing with this today so I might be seeing overlap?  Let me bang on this config some more, tx

bkaufx, 2025-03-21:
The names don't really matter at all.  I wouldn't worry about that.  You just have to click on the main light entity in home assistant.  It will pop up a little control window with RGB/temp modes, sliders to control, and a drop-down for effects all in one little popup.  That drop-down for effects is what you are looking for.

It will look like this.  See the effect at the bottom ("solid").

[image]

davlaw, 2025-03-21:
well fudge, that was easy enough, just got my eye to look at the right place

Consider "Solved"

closed 2025-03-21 by davlaw

davlaw, 2025-03-21:
I did find them here in the gui

Keep looking at the effect field..  

[image]